ALIGN-analoglayout/ALIGN-public #574: Define schema for design_info and user_constraints

soneryaldiz, 2021-03-16:
- [ ] Rename design_info as layout_config and define a schema. Remove vt_type from this schema (this info exists in netlist).
- [ ] Adopt layout_config schema 
- [ ] Define a schema for constraint_user
- [ ] Adopt constraint_user schema. Remove constraint_user from layers.json

soneryaldiz, 2021-03-16:
@kkunal1408 , do you want to co-develop this? I can start a PR to define the schemas and you can use these schemas after reviewed by @parijatm .

kkunal1408, 2021-03-16:
Sure, we can work on it after tomorrow afternoon. I have an exam tomorrow.

Kunal

soneryaldiz, 2021-03-16:
I did not mean to rush at all. Good luck with the exam! :)

kkunal1408, 2021-03-24:
Let me know when do you want to discuss this?

soneryaldiz, 2021-04-27:
Closing this issue as this has been / being addressed in other issues.

closed 2021-04-27 by soneryaldiz