ALIGN-analoglayout/ALIGN-public #608: Constraint DSL: Mirror

soneryaldiz, 2021-04-08:
Provide a constraint that can `mirror` a given list of `instances` along the specified `axis`

```python
class Mirror(ConstraintBase):
    ''' 
        Mirror `instances` along `x_axis` and/or `y_axis`
    '''
    name: Literal['mirror']
    instances: List[str]
    axis: Optional[Literal['x', 'y', 'xy']]
 ```

soneryaldiz, 2021-04-08:
@parijatm , please review at your convenience.

854768750, 2021-04-22:
@soneryaldiz This can also be implemented. Is it going to place 2 blocks like this?
[image]

soneryaldiz, 2021-04-27:
Not quite. The idea is to place only the flipped version of an instance. 
```
xi0 ip op1 comparator
xi1 ip op2 comparator
```

```java
{ "constraint": "Mirror", "instances": ["xi0"], "axis": "x"}
```
xi0 is flipped when placed on the canvas.

This enhancement is lower priority and can be addressed later.

closed 2022-01-18 by soneryaldiz