ALIGN-analoglayout/ALIGN-public #612: Constraint DSL: Bound

soneryaldiz, 2021-04-08:
Provide a way to constrain bounding box of a circuit.

```python
class Bound(ConstraintBase):
    ''' 
        Define boundary for a subcircuit
        aspect_ratio = height / width
    '''
    name: Literal['bound']
    aspect_ratio_min: Optional[float] = 0.75
    aspect_ratio_max: Optional[float] = 1.25
    max_height: Optional[int] = None
    max_width: Optional[int] = None
```

854768750, 2021-04-22:
This constraint is ready in #652 .

soneryaldiz, 2021-04-23:
- [ ] Translate Bound to Aspect_Ratio and Boundary.
- [ ] Add checker for Bound.
Aspect Ratio = weight / height
Examples:
{ "const_name": "Aspect_Ratio", "weight": 1, "ratio_low": 0.25, "ratio_high":4.0 }
{ "const_name": "Boundary", "max_width": 8, "max_height": 8 }

854768750, 2021-04-23:
> * [ ]  Translate Bound to Aspect_Ratio and Boundary.
> * [ ]  Add checker for Bound.
>   Aspect Ratio = weight / height
>   Examples:
>   { "const_name": "Aspect_Ratio", "weight": 1, "ratio_low": 0.25, "ratio_high":4.0 }
>   { "const_name": "Boundary", "max_width": 8, "max_height": 8 } where height and width are in micrometers.

Aspect Ratio = width / height

soneryaldiz, 2021-04-23:
Typo corrected :)

soneryaldiz, 2022-01-14:
This issue is complete: Boundary, AspectRatio constraints implemented.

closed 2022-01-14 by soneryaldiz