ALIGN-analoglayout/ALIGN-public #620: Current flow doesn't run correctly with number of variants greater than one and the design is hierarchical

stevenmburns, 2021-04-13:
@Lastdayends @854768750 This can be reproduced by:

```bash
mkdir -p work/cascode_current_mirror_ota
cd work/cascode_current_mirrot_ota
schematic2layout.py $ALIGN_HOME/examples/cascode_current_mirror_ota -c -n 2
```

Here is the tail of the log:
```text
...
PnR.cap_placer.Placer_Router_Cap.Placer_Router_Cap INFO : Begin CC Capacitor Placement and Router
PnR.cap_placer.Placer_Router_Cap.Placer_Router_Cap INFO : End CC Capacitor Placement and Router
align.pnr.toplevel WARNING : Placer did not provide numLayout (2 > 1) layouts
PnR.PnRDB.PnRdatabase.CheckinHierNode INFO : CheckinHierNode
align.pnr.toplevel INFO : Topo order: 0 cascode_current_mirror_ota
python3.8: ../../../PlaceRouteHierFlow/PnRDB/PnRdatabase.cpp:1932: void PnRdatabase::AddingPowerPins(PnRDB::hierNode&): Assertion `0 <= iter && iter < node.Blocks[iter2].instance[k].dummy_power_pin.size()' failed.
Aborted
```

854768750, 2021-04-14:
I will see how to solve this.

stevenmburns, 2021-04-14:
Great. Make sure you start from the latest master.

854768750, 2021-04-14:
@stevenmburns  I think you have solved this in https://github.com/ALIGN-analoglayout/ALIGN-public/pull/618/commits/b7d9c29415a11caa347f681aa58e2b18f8a4b1e8

stevenmburns, 2021-04-14:
Thanks. I'll take a look.

stevenmburns, 2021-04-14:
@854768750 Looks like something I did fixed the core dump. Maybe you can instead working on making the placer return different placements. As far as I can tell, both `cascode_current_mirror_ota` and `switched_capacitory_filter` return the same placement.

stevenmburns, 2021-04-14:
@854768750 There also seems to be an issue with `cascode_current_mirror_ota` when you set `n` to 3.
```text
align.pnr.toplevel INFO : Topo order: 2 CASCODED_CMC_NMOS
align.pnr.toplevel WARNING : Placer did not provide numLayout (3 > 2) layouts
PnR.PnRDB.PnRdatabase.CheckinHierNode INFO : CheckinHierNode
PnR.PnRDB.PnRdatabase.CheckinHierNode INFO : CheckinHierNode
align.pnr.toplevel INFO : Topo order: 0 cascode_current_mirror_ota
PnR.PnRDB.PnRdatabase.CheckinHierNode INFO : CheckinHierNode
PnR.PnRDB.PnRdatabase.CheckinHierNode INFO : CheckinHierNode
PnR.PnRDB.PnRdatabase.CheckinHierNode INFO : CheckinHierNode
PnR.router.Router.RouteWork INFO : GcellGlobalRouter: CASCODED_CMC_PMOS
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Format Issue 
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Please check the net vdd in module CASCODED_CMC_PMOS
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Especial the pin S in subblock M3
python3.8: ../../../PlaceRouteHierFlow/router/GcellGlobalRouter.cpp:567: GcellGlobalRouter::GcellGlobalRouter(PnRDB::hierNode&, PnRDB::Drc_info&, int, int, const string&): Assertion `0' failed.
Aborted
```

stevenmburns, 2021-04-14:
@854768750 We can also try to make it do something reasonable for `telescopic_ota` (no hierarchy.) It seems to generate different placements as n is increased (the ones produced when n = 2 does not include the one produced with n =1).

854768750, 2021-04-15:
I will see how to solve it.

854768750, 2021-04-15:
[image]
@stevenmburns It is now able to generate 2 different placements.

stevenmburns, 2021-04-16:
The two different layouts look great.
I introduced a bug that is solved by #630 that probably is the reason you didn't see the original coredump. After merging that in, you might need to look at this again.

854768750, 2021-04-19:
this issue is solved

stevenmburns, 2021-04-19:
Solved by #624

closed 2021-04-19 by stevenmburns