ALIGN-analoglayout/ALIGN-public #622: Include the guardrings optionally in the schematic2layout.py flow

stevenmburns, 2021-04-14:
We also need tests for this.

stevenmburns, 2021-04-14:
@Lastdayends @854768750 How do we use guardrings now? Is there a way to do this through schematic2layout.py or do we recompile the code?

Lastdayends, 2021-04-14:
@stevenmburns 

I think we can not do guard ring through schematic2layout.py. 

Guardrings needs additional primitives (can not generate by ALIGN automatically). Such primitives is stored in the testcase_guarding.

Currently, guardrings can only be done with old flow like:
./pnr_compiler ./testcase_guardings testcase_guardings.v ...

Lastdayends, 2021-04-14:
@arvuce22 

Just to double check, the primitive of guardings can not be generated automatically by ALIGN.

arvuce22, 2021-04-15:
@Yaguang Li ***@***.***>
I will fix this.

kkunal1408, 2021-04-27:
Automation added in-branch #649

closed 2021-04-27 by kkunal1408