ALIGN-analoglayout/ALIGN-public #626: Printing DRC numbers on CI reports

kkunal1408, 2021-04-15:
We have a max bound on the number of DRCs. This max bound masks any change in the DRC numbers within this range. Can we print these DRCs numbers which can be viewed visually? Later we can create some summary reporting methods for these.

stevenmburns, 2021-04-19:
We can run a script over the result directories to get this information. This information is stored in the 3_pnr/*.errors files. We could add this to the artifact directories of our CI run.

stevenmburns, 2021-04-23:
I like my suggestion better. I’ll build a ‘pandas’ script that will generate the number of errors of each kind by looking at the resulting .errors files and produce a .csv file with all the information in it.

soneryaldiz, 2021-04-23:
Can we set max errors tolerated to zero for select (small and fundamental) circuit blocks? There should not be any excuse for opens or shorts for a five/six transistor OTA or a comparator. I started adding such tests for align/pdk/finfet under tests/pdk/finfet/test_examples.py

parijatm, 2021-04-23:
Yes. Just make make the appropriate changes to test-integration in config.yml. Or add another job to the workflow. You will see --maxerrors=20 explicitly in the test-integration pytest command.

By the way, if you have tagged tests/pdk/finfet/test_examples.py with the nightly flag, they don't get executed at the moment. The only nightly tasks that get run are tests/**integration** & tests/**pdks**

closed 2021-11-08 by kkunal1408