ALIGN-analoglayout/ALIGN-public #627: Study non-determinism in layout generation

parijatm, 2021-04-15:
I have anecdotally noticed differing numbers of DRC errors in different runs.

In general, we should be generating the same layout with the same initial seed unless we have a good reason not to.

I propose we take a set of candidate circuits and run ALIGN on it a few hundred times. And check to make sure it generates the same layout  every time. If not, we should examine where the non-determinism is coming from and whether we are okay with that long term (most users expect to get the same layout every time unless they are changing the stimuli).

Lastdayends, 2021-04-15:
@kkunal1408 

Just want to double check, if the constraints  are deterministic.

kkunal1408, 2021-04-15:
@Lastdayends I have some checks for some design where I compare with a golden constraint generated dataset. If we are looking into a specific design, let me know  I will double-check and put that design in regression.

854768750, 2021-04-17:
I also find this problem when I am running the entire flow of `switched_capacitor_filter` on the master branch.

kkunal1408, 2021-04-17:
I will check if there are any changes across runs on SCF and put the results under pytest.

854768750, 2021-04-17:
It seems the latest master branch does not have this problem.

kkunal1408, 2021-04-27:
A test to compare SCF constraint with golden has been added in branch #649 . Will close this for now and open it again if we face the issue again.

closed 2021-04-27 by kkunal1408