ALIGN-analoglayout/ALIGN-public #628: Power connectivity is broken

soneryaldiz, 2021-04-15:
How to reproduce: Run high_speed_comparator on master branch:
```schematic2layout.py $ALIGN_HOME/examples/high_speed_comparator -p $ALIGN_HOME/pdks/FinFET14nm_Mock_PDK -c```

Power connectivity is broken, see the snapshot below:
1. Notice supply0 and supply1
2. Notice mn1_mn2/vss, mn3_mn4/vss,...

Unfortunately, none of the integration tests catch this problem.

[image]

soneryaldiz, 2021-04-15:
@stevenmburns , could this be related to your recent refactoring for *.verilog.json?

stevenmburns, 2021-04-16:
Maybe. I’ll take a look. We need actual pass/fail for all functionality we want to preserve.

stevenmburns, 2021-04-19:
Solved by #630 We still need to add decent tests so this doesn't happen again.

closed 2021-04-19 by stevenmburns