ALIGN-analoglayout/ALIGN-public #647: Constraint DSL: Generate

soneryaldiz, 2021-04-22:
Provide a constraint that can generate a customized primitive for a given list of instances. 
These instances should be excluded from subcircuit identification. 
Parameters should be passed to the generating class.

```python
class Generate():
    instances: List[str]
    name: str
    interleave: Optional[Literal["common_centroid", "interdigitated"]] = "common_centroid"
    parameters: Optional[Dict[str, Union[str,int]]
```

Example:
```json
{
  "constraint": "generate",
  "instances": ["mn1", "mn2"],
  "name": "diffpair",
  "parameters": {
    "n_rows": 4,
    "add_guardring": true
  }  
}
```

soneryaldiz, 2021-04-22:
@kkunal1408 , could you please review this enhancement? This is to customize transistor, passive, unknown device primitives.

kkunal1408, 2021-04-22:
Looks fine to me. Do we want more configuration parameters apart from a number of rows? e.g., dummy, n_columns, stylr=common_centroid and interdigitated.

soneryaldiz, 2021-04-22:
Let's start with bare minimum in the schema that is `instances, `name` and `interleave` and maybe `add_guardring`.  Let's have everything else optional for now.

kkunal1408, 2021-04-22:
okay. I will add this constraint. I think we have capabilities available for this. Based on current implementation is it fine if we ask user to give the name the same as in the library i.e., DP_NMOS/DP_PMOS, or do you want it to be handled internally?

soneryaldiz, 2021-04-22:
No, there is no need to ask user nor MOSGenerator needs to know about it. All compiler has to do is to call the associated primitive (aka array) generator which could be MOSGenerator, CAPGenerator, XGenerator, etc. Please see this #644 . It is possible to generalize this to arbitrary subcircuits but we can keep the scope to device arrays for now. Maybe, we should call this ArrayDevices :-)

kkunal1408, 2021-04-22:
Can we discuss this during today meeting?

kkunal1408, 2022-03-10:
From other discussion:
	templates.sp: * @: Generator(name="mos", params=[])
	const.json:   * @: Generator(instances=["mn0"], params=[insert_dummy, share_diff])
	const.json:   * @: Generator(instances=["mn1", "mn2"], params=[nrow, id, sparse])
		combine template and const where const has precedence

	What happens with a generator constraint?
		Generator(instances=["mn1", "mn2"], params=[nrow, id, sparse])
			create a virtual subcircuit guid_x for mn1 and mn2
			make sure mn1 and mn2 to matches to a template in templates.sp
			.subcircuit guid_x:
				const: generator(name="mos", params=[nrow, id, sparse])

soneryaldiz, 2022-03-11:
@kkunal1408 , is there a question?

kkunal1408, 2022-03-11:
No, I copy-pasted the future work which we identified from #964 . I think both have some relation. We can brainstorm both together.

soneryaldiz, 2022-05-15:
This issue indirectly addressed in recent PR's

closed 2022-05-15 by soneryaldiz