ALIGN-analoglayout/ALIGN-public #650: Fixed height cell

kkunal1408, 2021-04-22:
Put an example for fixed height cells. The requirement is for mixed-signal designs. Get help from Meghna to get SAR_ADC test case and pass this down to Arvind.

meghna09, 2021-12-02:
Can we move the cell height parameter to the layers.json or somewhere the user has access to change it? I think the cell generator is able to support the requirement of having a specified height while respecting the no. of fins per unit cell mentioned in the netlist.

arvuce22, 2022-01-20:
@kkunal1408 We have fixed most of it in #870.  Do we need "unit_size_nmos" and "unit_size_pmos" values?

kkunal1408, 2022-01-20:
@arvuce22 , @meghna We would need this to check if nfin defined in netlist < unit_size_mos. We can rename this to something which would make more sense.

arvuce22, 2022-01-20:
@kkunal1408 @meghna09 Currently, we have four parameters in 'layers.json': "unit_size_nmos", "max_size_nmos", "max_size_pmos", "unit_size_pmos". Instead of these we can have two parameters "unit_size_mos" and "max_size_mos"

meghna09, 2022-01-20:
@arvuce22  max_size_mos would represent the cell height? What does unit_size_mos imply?

arvuce22, 2022-01-20:
@meghna09 We will remove "cell height", and "max_size_mos" will define maximum number of fins in a transistor (this is a limit from pdk)

meghna09, 2022-01-21:
@arvuce22 is having cell height and max_size_mos (limit from pdk) more suitable than unit_size_mos? We can discuss this in our meeting.

parijatm, 2022-01-21:
@arvuce22 to lead discussion on this

closed 2022-01-22 by arvuce22