ALIGN-analoglayout/ALIGN-public #656: [Question] How do we build the C++ code in Release mode?

stevenmburns, 2021-04-23:
@parijatm There are two modes of development work on Align. One, where you are debugging C++ code and want optimization turned off and debug symbols added to the executable. In the other mode, you are working most in python and want the C++ code to run as fast as possible. Currently I'm hacking the setup.py file to enable Release mode, doing something like this:
```text
diff --git a/setup.py b/setup.py
index 560d2a777..a83f0ee96 100644
--- a/setup.py
+++ b/setup.py
@@ -36,8 +36,9 @@ def align_manifest_filter(cmake_manifest):
 devmode = 'develop' in sys.argv
 # if devmode and not any(x.startswith('-DBUILD_TESTING') for x in sys.argv):
 #     cmake_args.append('-DBUILD_TESTING=ON')
-if devmode and '--build-type' not in sys.argv:
+if False and devmode and '--build-type' not in sys.argv:
     sys.argv.extend(["--build-type", "Debug"])
+sys.argv.extend(["--build-type", "Release"])
 
 setup(name='align',
       version=version,
```
What do you think I should be doing instead?

parijatm, 2021-04-23:
I would _like_ to do it just like BUILD_TESTING was done before:
```python
 devmode = 'develop' in sys.argv
if devmode and not any(x.startswith('-DBUILD_TESTING') for x in sys.argv):
     cmake_args.append('-DBUILD_TESTING=ON')
if devmode and not any(x.startswith('--build-type') for x in sys.argv):
     sys.argv.extend(['--build-type', 'Debug'])
```
And then use `--install-option='--build-type=Release'` whenever we want to force Release mode (or vice versa when not in devmode).

However, pip behaves somewhat differently with and without the --install-option (It tries build all the dependencies from source if I remember right). The way I handled it when using --install-option for CODE_COVERAGE is as follows:
```console
# Install align with dependencies
pip install -v --no-build-isolation -e .[test]
# Rebuild align alone with appropriate flags
pip install -v --no-build-isolation -e . --no-deps --install-option="-DCODE_COVERAGE=ON"
```
This can be optimized a little by installing just the dependencies in step 1 but the commands get a bit harder to understand.

The other way to do it is to skip pip altogether and use setup.py directly
```console
python setup.py develop -DCODE_COVERAGE=ON
```
This feels cleaner but I haven't had time to investigate what issues we are introducing by circumventing pip.

stevenmburns, 2021-04-24:
Resolved in #660 
See README.md for what to do, similar to above.

closed 2021-04-24 by stevenmburns