ALIGN-analoglayout/ALIGN-public #664: PnR does not appear to respect Ordering constraint

parijatm, 2021-04-25:
#658 appears to have broken Vertical ordering constraint in PnR.

Reproduce using: 
pytest -vv --runnightly tests/integration/test_all.py::test_A[20-FinFET14nm_Mock_PDK-ring_oscillator]

Relevant log messages:
```console
INFO     align.pnr.write_constraint:write_constraint.py:26 input constraints __root__=[HorizontalDistance(constraint='horizontal_distance', abs_distance=0), VerticalDistance(constraint='vertical_distance', abs_distance=0), Order(constraint='order', instances=List[str](__root__=['M1', 'M0']), direction='top_to_bottom', abut=False)]
INFO     align.pnr.write_constraint:write_constraint.py:172 Const mapped to PnR const format [{'const_name': 'bias_Hgraph', 'distance': 0}, {'const_name': 'bias_Vgraph', 'distance': 0}, {'direction': 'V', 'const_name': 'Ordering', 'blocks': ['M1', 'M0']}]
...
ERROR    align.schema.constraint:constraint.py:599 Failed to add constraint constraint='set_bounding_box' instance='M1' llx=0 lly=0 urx=1280 ury=3696 due to conflict with one or more of:
ERROR    align.schema.constraint:constraint.py:602 constraint='order' instances=List[str](__root__=['M1', 'M0']) direction='top_to_bottom' abut=False
ERROR    align.schema.constraint:constraint.py:602 constraint='set_bounding_box' instance='M0' llx=1280 lly=0 urx=2560 ury=3696
```

It looks like the 'V' flag is being ignored and it is generating a horizontal ordering instead?

stevenmburns, 2021-04-25:
This is a better command line for me:
```bash
pytest -vv --runnightly -k 'ring_oscillator' -- tests/integration/test_all.py 
```

stevenmburns, 2021-04-25:
@parijatm @srini229 I suspect this only has to do with changing some of the placer hyperparameters. The enumerative placer isn't being called on this 'large' testcase. What was changed was the simulated annealing ALPHA and I suspect that this uncovered a latent bug. I'll investigate a little bit more.

stevenmburns, 2021-04-25:
Changing ALPHA back to 0.95 doesn't seem to fix this issue so maybe my conjecture is wrong.

srini229, 2021-04-25:
I had missed calling the KeepOrdering routine during the enumeration. I fixed it in another checkin

854768750, 2021-04-26:
This is solved. Should this be closed?

parijatm, 2021-04-26:
Yes. Closing. Fixed by #663 and #666.

closed 2021-04-26 by parijatm