ALIGN-analoglayout/ALIGN-public #673: Merge to master CI failing

stevenmburns, 2021-04-28:
It seems that 'BUFFER_VCM' is failing sometimes
To (try to) reproduce locally:
```
schematic2layout.py /home/smburns/DARPA/ALIGN-public/examples/*/BUFFER_VCM -c
```
In one run I got:
```
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/work/BUFFER_VCM$ schematic2layout.py /home/smburns/DARPA/ALIGN-public/examples/*/BUFFER_VCM -c
align.main INFO : Steps to run in the flow: ['1_topology', '2_primitives', '3_pnr'][0:3] => ['1_topology', '2_primitives', '3_pnr']
align.main INFO : READ file: /home/smburns/DARPA/ALIGN-public/examples/modified_USC_UW_UT_testcases/BUFFER_VCM/BUFFER_VCM.sp subckt=BUFFER_VCM, flat=0
align.compiler.compiler INFO : Starting topology identification...
align.compiler.user_const INFO : No user constraints found for block BUFFER_VCM in path /home/smburns/DARPA/ALIGN-public/examples/modified_USC_UW_UT_testcases/BUFFER_VCM
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w7 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w7 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w6 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w7 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w9 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w9 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w9 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w9 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w4 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w4 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w4 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w4 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter w0 not defined. Using value=unit_size. Please fix netlist
align.compiler.merge_nodes ERROR : Parameter l0 not defined. Using value=unit_size. Please fix netlist
[... 81 lines cut ...]
3_pnr/Results/Cap_10f_AspectRatio_1x1.json-        0,
3_pnr/Results/Cap_10f_AspectRatio_1x1.json-        0,
3_pnr/Results/Cap_10f_AspectRatio_1x1.json-        32000,
3_pnr/Results/Cap_10f_AspectRatio_1x1.json-        -1303000112
3_pnr/Results/Cap_10f_AspectRatio_1x1.json-    ],
--
3_pnr/Results/Cap_50f_AspectRatio_2x3.json-{
3_pnr/Results/Cap_50f_AspectRatio_2x3.json:    "bbox": [
3_pnr/Results/Cap_50f_AspectRatio_2x3.json-        0,
3_pnr/Results/Cap_50f_AspectRatio_2x3.json-        0,
3_pnr/Results/Cap_50f_AspectRatio_2x3.json-        60000,
3_pnr/Results/Cap_50f_AspectRatio_2x3.json-        -1303022792
3_pnr/Results/Cap_50f_AspectRatio_2x3.json-    ],
--
3_pnr/Results/Cap_50f_AspectRatio_3x2.json-{
3_pnr/Results/Cap_50f_AspectRatio_3x2.json:    "bbox": [
3_pnr/Results/Cap_50f_AspectRatio_3x2.json-        0,
3_pnr/Results/Cap_50f_AspectRatio_3x2.json-        0,
3_pnr/Results/Cap_50f_AspectRatio_3x2.json-        92800,
3_pnr/Results/Cap_50f_AspectRatio_3x2.json-        -1302998432
3_pnr/Results/Cap_50f_AspectRatio_3x2.json-    ],
```

I ran this a few more times and didn't get the same error.
Maybe we have an uninitialized variable somewhere. `valgrind` might be a good way to check.

stevenmburns, 2021-04-28:
@854768750 @Lastdayends It seems that valgrind finds an illegal read issue in the cap placer:
```bash
valgrind -- schematic2layout.py /home/smburns/DARPA/ALIGN-public/examples/*/BUFFER_VCM -c
```
There is a lot of Python related stuff but then this:
```
==23510== Invalid read of size 4
==23510==    at 0x2A3CFC19: PnRDB::point::operator-(PnRDB::point const&) const (datatype.h:73)
==23510==    by 0x2A5D7FF0: Placer_Router_Cap::GetPhysicalInfo_merged_net(std::vector<Placer_Router_Cap::net, std::allocator<Placer_Router_Cap::net> >&, std::vector<int, std::allocator<int> >&, PnRDB::Drc_info const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int, int) (capplacer.cpp:1278)
==23510==    by 0x2A5D8CD6: Placer_Router_Cap::GetPhysicalInfo_router(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, PnRDB::Drc_info const&) (capplacer.cpp:1432)
==23510==    by 0x2A5D07B8: Placer_Router_Cap::Placer_Router_Cap_function(std::vector<int, std::allocator<int> >&, std::vector<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, int, int, PnRDB::Drc_info const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::lefMacro, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, PnRDB::lefMacro> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (capplacer.cpp:357)
==23510==    by 0x2A5DDEBE: Placer_Router_Cap::Common_centroid_capacitor_aspect_ratio(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, PnRDB::hierNode&, PnRDB::Drc_info&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::lefMacro, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, PnRDB::lefMacro> > > const&, bool, int) (capplacer.cpp:1929)
==23510==    by 0x2A5CEA22: Placer_Router_Cap::Placer_Router_Cap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, PnRDB::hierNode&, PnRDB::Drc_info&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::lefMacro, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, PnRDB::lefMacro> > > const&, bool, int) (capplacer.cpp:63)
==23510==    by 0x2A5CD6DF: Placer_Router_Cap_Ifc::Placer_Router_Cap_Ifc(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, PnRDB::hierNode&, PnRDB::Drc_info&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::lefMacro, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, PnRDB::lefMacro> > > const&, bool, int) (CapPlacerIfc.cpp:7)
==23510==    by 0x2A2E6201: Placer_Router_Cap_Ifc* pybind11::detail::initimpl::construct_or_initialize<Placer_Router_Cap_Ifc, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::hierNode&, PnRDB::Drc_info&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::lefMacro, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, PnRDB::lefMacro> > >&, bool, int, 0>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, PnRDB::hierNode&, PnRDB::Drc_info&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::lefMacro, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char [... cut ...]  >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::hierNode&, PnRDB::Drc_info&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, PnRDB::lefMacro, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, PnRDB::lefMacro> > >&, bool, int) const (init.h:175)
==23510== 
```

854768750, 2021-04-29:
I am trying to try this. Where is BUFFER_VCM? It is not in examples/ folder.

srini229, 2021-04-29:
ALIGN-public/examples/modified_USC_UW_UT_testcases/BUFFER_VCM/

stevenmburns, 2021-05-06:
@srini229 Maybe you figured this out already but if you want to eliminate most of the valgrind errors on python related code you can invoke it like this:
```
PYTHONMALLOC=malloc valgrind -- schematic2layout.py $ALIGN_HOME/examples/*/BUFFER_VCM -c
```

Lastdayends, 2021-05-10:
Has been fixed in branch #685.