ALIGN-analoglayout/ALIGN-public #679: CI Failure: telescopic_ota_guard_ring has 57 DRC errors

parijatm, 2021-04-29:
```console
FAILED tests/integration/test_all.py::test_A[20-FinFET14nm_Mock_PDK-telescopic_ota_guard_ring]
```
Appears to be caused by #649

stevenmburns, 2021-04-30:
@parijatm @arvuce22 Can we just take this out of the tests (both checkin and master merge) until we get it fixed. Seems like we need to do some work with either the primitive itself or the on-grid checking code.

parijatm, 2021-04-30:
Certainly @stevenmburns

parijatm, 2021-04-30:
Please just skip it in whatever PR you are about to merge @stevenmburns . That way @kkunal1408 and @arvuce22 can just go back in and reenable it when they get it working.

854768750, 2021-04-30:
How can I run telescopic_ota_guard_ring using schemetic2layout.py

stevenmburns, 2021-04-30:
Should just be:
```
schematic2layout.py $ALIGN_HOME/examples/telescopic_ota_guard_ring -c
```

stevenmburns, 2021-04-30:
Or if you want to use pytest:
```
pytest --runnightly -vv -k telescopic_ota_guard_ring
```

854768750, 2021-04-30:
> Should just be:
> 
> ```
> schematic2layout.py $ALIGN_HOME/examples/telescopic_ota_guard_ring -c
> ```

It is not in examples folder.

stevenmburns, 2021-04-30:
Update to master. I should be there.

854768750, 2021-04-30:
One drc is `DRC ERROR MinLength violation on M1: vss[1024, 300, 1056, 372]`
[image]
It says the routing metal is too short. But it is already connected to the guradring M1 metal. Is there any way to solve this?

854768750, 2021-04-30:
Some other drc and opens are caused by offgrid. I am looking into them.

854768750, 2021-04-30:
Now telescopic_ota_guard_ring has drc/lvs, it is the minlength mentioned above. Offgrid is solved in #683

854768750, 2021-05-14:
This issue is solved. Can we close it?

stevenmburns, 2021-05-14:
Down to 1 probably false DRV. Looks good.

closed 2021-05-14 by stevenmburns