ALIGN-analoglayout/ALIGN-public #695: Various blocks excluded from CI on merge to master because of failures

stevenmburns, 2021-05-04:
@kkunal1408 We are using the following string now to block running problematic designs in our merge to master CI
```
"not SAR and not DLL and not single_to_differential_converter and not COMPARATOR_2LEVEL_BIDIRECTIONAL_MAC_SKEW and not CTD and not BUFFER_VREFP and not BUFFER_VREFP2 and not CTDSM_CORE_NEW and not BUFFER_VCM and not vco_dtype_12_hierarchical and not Sanitized_Coarse_Comp_CK and not COMP_GM_STAGE_0415 and not linear_equalizer and not test_vga and not Gm1_v5_Practice and not powertrain_binary and not SW_Cres_v3_5 and not ldo_error_amp and not telescopic_ota_guard_ring"
```

We should go though these one by one but of particular note is:
```
powertrain_binary
```
Both powertrain and powertrain_thermo seem to run okay. This block generates an empty primitivies list.
Maybe it is something to start on

parijatm, 2022-01-21:
Progress has been made but there are a few that still need to be looked at.

Excluded test list is now in test_integration.py

parijatm, 2022-01-21:
Note: If we don't want those examples, we should remove from examples directory altogether. The exclusion in the python file should be purely temporary while we work on them.

kkunal1408, 2022-03-10:
These examples have been removed. Will add the right test cases through PR #959

closed 2022-03-10 by kkunal1408