ALIGN-analoglayout/ALIGN-public #697: Installation issue  with univ of washington

kkunal1408, 2021-05-06:
Running setup.py develop for align
    ERROR: Command errored out with exit status 1:
     command: /net/gaylord/t/kamineni/ALIGN_installation_trial2/ALIGN-public/general/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/net/gaylord/t/kamineni/ALIGN_installation_trial2/ALIGN-public/setup.py'"'"'; __file__='"'"'/net/gaylord/t/kamineni/ALIGN_installation_trial2/ALIGN-public/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: /net/gaylord/t/kamineni/ALIGN_installation_trial2/ALIGN-public/
    Complete output (15 lines):
    /tmp/pip-build-env-0meokkn7/overlay/lib/python3.7/site-packages/cmake/data/bin/cmake: /lib64/libc.so.6: version `GLIBC_2.14' not found (required by /usr/um/gcc-7.1.0/lib64/libstdc++.so.6)
    /tmp/pip-build-env-0meokkn7/overlay/lib/python3.7/site-packages/cmake/data/bin/cmake: /lib64/libc.so.6: version `GLIBC_2.16' not found (required by /usr/um/gcc-7.1.0/lib64/libstdc++.so.6)
    /tmp/pip-build-env-0meokkn7/overlay/lib/python3.7/site-packages/cmake/data/bin/cmake: /lib64/libc.so.6: version `GLIBC_2.17' not found (required by /usr/um/gcc-7.1.0/lib64/libstdc++.so.6)
    /tmp/pip-build-env-0meokkn7/overlay/lib/python3.7/site-packages/cmake/data/bin/cmake: /lib64/libc.so.6: version `GLIBC_2.14' not found (required by /usr/um/gcc-7.1.0/lib64/libgcc_s.so.1)
    /usr/um/python-3.7/lib/python3.7/distutils/dist.py:274: UserWarning: Unknown distribution option: 'cmake_process_manifest_hook'
      warnings.warn(msg)
      File "/tmp/pip-build-env-0meokkn7/overlay/lib/python3.7/site-packages/skbuild/setuptools_wrap.py", line 560, in setup
        cmkr = cmaker.CMaker(cmake_executable)
      File "/tmp/pip-build-env-0meokkn7/overlay/lib/python3.7/site-packages/skbuild/cmaker.py", line 95, in __init__
        self.cmake_version = get_cmake_version(self.cmake_executable)
      File "/tmp/pip-build-env-0meokkn7/overlay/lib/python3.7/site-packages/skbuild/cmaker.py", line 82, in get_cmake_version
        "Problem with the CMake installation, aborting build. CMake executable is %s" % cmake_executable)
    Problem with the CMake installation, aborting build. CMake executable is cmake
    Traceback (most recent call last):
    
    ----------------------------------------
ERROR: Command errored out with exit status 1: /net/gaylord/t/kamineni/ALIGN_installation_trial2/ALIGN-public/general/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/net/gaylord/t/kamineni/ALIGN_installation_trial2/ALIGN-public/setup.py'"'"'; __file__='"'"'/net/gaylord/t/kamineni/ALIGN_installation_trial2/ALIGN-public/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.

kkunal1408, 2021-05-06:
cmake/2.8.8(default)
python3.7

parijatm, 2021-05-06:
Error seems to indicate that the libstdc++.so.6 has been compiled with a newer glibc (that is not provided by the default /lib64/libc.so.6).

Few options at this stage:
1. Upgrade the system GLIBC using apt / yum etc (probably not practical if this is a shared server)
2. Modify LD_LIBRARY_PATH to point to a more recent libc if one exists already
3. Don't try to install from source at all but use pre-built binaries 

For option 3, you will need to send them the tar archive generated by publish-wheel on master. Latest generated wheel directory appears to be [here](https://39048-177476440-gh.circle-artifacts.com/0/export/align-wheelhouse-f6aeddc.tar.gz). Instructions for wheel based installation can be found in the last comment in #572

parijatm, 2021-05-06:
My availability is extremely limited until 5/24. Please tag @stevenmburns @soneryaldiz if you need additional help.

kkunal1408, 2021-05-14:
Thanks Parijat.
option 2 worked by setting LD_LIBRARY_PATH: setenv LD_LIBRARY_PATH /usr/lib/libc.so.6:$LD_LIBRARY_PATH
option 3 works but if I update python files locally it does not get reflected. Reinstallation also does not work in this case as python files are also picked from wheel. To make it work we had to overwrite files in the installed directory.

closed 2021-05-14 by kkunal1408

parijatm, 2021-05-14:
Yes installation from wheel does not allow development mode. Glad it got resolved!