ALIGN-analoglayout/ALIGN-public #705: Is it possible to copy new constraint directly to PnR input?

854768750, 2021-05-12:
@soneryaldiz 
We want to add some constraints while they are not supported by constraint translator yet. So errors occur.
Is it possible to directly copy some constraints like
```
    {
        "key1": item1,
        "key2": item2,
        "constraint": "new_constraint1"
    }
```
to PnR input folder like
```
    {
        "key1": item1,
        "key2": item2,
        "const_name": "new_constraint1"
    }
```

soneryaldiz, 2021-05-13:
As creating a new soft constraint is quite straightforward, I would suggest not to open this backdoor or bypass channel. Let's discuss in team meeting further if this is a significant slow down.

854768750, 2021-05-14:
Solved by #709

closed 2021-05-14 by 854768750