ALIGN-analoglayout/ALIGN-public #716: Question: Is there a constraint to tell the placer to choose the same leaf cell for two or more different instances?

stevenmburns, 2021-05-21:
@Lastdayends @854768750 @parijatm @soneryaldiz We now have this concept of abstract_template_name (before placement is solidified) and concrete_template_name (after placement as been done) for each template that we instantiate. In the placer, the gdsFile field specifies the concrete_template_name for leaf cells. The selectedInstance field specifies this (along with the master name) for internal cells in hierarchy.

Do we have a constraint to force this to be the same for a set of instances? Something like:
```json
{
   "constraint": "SameConcreteName",
   "instances": ["i0", "i1", "i2", "i3']
}
```

854768750, 2021-05-21:
Does this mean "i0", "i1", "i2", "i3' have the same selectedInstance value?

stevenmburns, 2021-05-21:
Yes, for internal cells in the hierarchy the selectedInstance value needs to be the same as well as the master name. For leaf cells they should have the gdsFile name.
Here is some Python code that is computing these names from the DB model and translating them into our placement_verilog_json file:
```python
    def traverse( hN, sel):
        for blk in hN.Blocks:
            child_idx = blk.child
            inst = blk.instance[blk.selectedInstance]

            b = inst.originBox
            new_r = b.LL.x, b.LL.y, b.UR.x, b.UR.y

            abstract_template_name = f'{inst.master}'
            if child_idx >= 0:
                concrete_template_name = f'{inst.master}_{blk.selectedInstance}'
                if concrete_template_name not in used_internal[abstract_template_name]:
                    new_hN = DB.CheckoutHierNode(child_idx, blk.selectedInstance)
                    traverse(new_hN, blk.selectedInstance)
                    used_internal[abstract_template_name][concrete_template_name] = (child_idx,blk.selectedInstance,new_r)
                else:
                    assert used_internal[abstract_template_name][concrete_template_name] == (child_idx,blk.selectedInstance,new_r)
            else:
                concrete_template_name = pathlib.Path(inst.gdsFile).stem
                if concrete_template_name not in used_leaves[abstract_template_name]:                
                    used_leaves[abstract_template_name][concrete_template_name] = new_r
                else:
                    assert used_leaves[abstract_template_name][concrete_template_name] == new_r
```
If `blk.child` is non-negative then the concrete template name is computed from as:
```python
                concrete_template_name = f'{inst.master}_{blk.selectedInstance}'
```
For leaves, `blk.child` is negative, then the concrete leaf name is (.stem strips of the directory and file extension):
```python
                concrete_template_name = pathlib.Path(inst.gdsFile).stem
```

854768750, 2021-05-21:
Currently we do not have this constraint. But we can add it.

stevenmburns, 2021-05-21:
It would be great if you could add this. You are probably doing something similar in a symmetry constraint, so maybe you can reuse some of that code.

The case I'm interesting is an array of current-starved inverters for a VCO. I don't want the layout for these to be different.

854768750, 2021-05-21:
Can you generate a testcase which has this constraint?

stevenmburns, 2021-05-21:
Sure. I've been using `vco_dtype_12_hierarchical_res_constrained` where I've added these constraints to one of the subhierarchies:
```json
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/work/high_speed_comparator$ cat ../../examples/vco_dtype_12_hierarchical_res_constrained/VCO_type2_65.const.json
[
    {"constraint": "GroupBlocks", "instances": ["I1<1>", "I1<2>", "I1<3>", "I1<4>"], "name": "top"},
    {"constraint": "GroupBlocks", "instances": ["I1<8>", "I1<7>", "I1<6>", "I1<5>"], "name": "bot"},
    {
	"constraint": "Order",
	"direction" : "left_to_right",
	"instances": ["I1<1>", "I1<2>", "I1<3>", "I1<4>"]
    },
    {
	"constraint": "Order",
	"direction" : "left_to_right",
	"instances": ["I1<8>", "I1<7>", "I1<6>", "I1<5>"]
    },
    {
	"constraint": "Order",
	"direction" : "top_to_bottom",
	"instances": ["top", "bot"]
    }
]
```

You'll want `["I1<1>", "I1<2>", "I1<3>", "I1<4>"]` to be the same as well as `["I1<8>", "I1<7>", "I1<6>", "I1<5>"]` (ideally all should be the same, but four at a time will work for now.

I'm guessing your are going to make up a PnR based constraint for this. I'll then make one in topology space and translate it to your PnR version.

You'll probably want to add the constraint to here (and also the similar file `bot.pnr.const.json`):
```json
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/work/vco_dtype_12_hierarchical_res_constrained$ cat 3_pnr/inputs/top.pnr.const.json 
{
    "constraints": [
        {
            "direction": "H",
            "const_name": "Ordering",
            "blocks": [
                "I1<1>",
                "I1<2>",
                "I1<3>",
                "I1<4>"
            ]
        },
        {
            "net1": {
                "name": "o<2>",
                "blocks": [
                    {
                        "type": "pin",
                        "name": "I1<1>",
                        "pin": "VOUT"
                    },
                    {
                        "type": "pin",
                        "name": "I1<2>",
                        "pin": "VIN"
                    },
                    {
                        "type": "terminal",
                        "name": "o<2>",
                        "pin": null
                    }
                ]
            },
            "net2": {
                "name": "o<3>",
                "blocks": [
                    {
                        "type": "pin",
                        "name": "I1<2>",
                        "pin": "VOUT"
                    },
                    {
                        "type": "pin",
                        "name": "I1<3>",
                        "pin": "VIN"
                    },
                    {
                        "type": "terminal",
                        "name": "o<3>",
                        "pin": null
                    }
                ]
            },
            "const_name": "SymmNet",
            "axis_dir": "V"
        },
        {
            "pairs": [
                {
                    "type": "sympair",
                    "block1": "I1<1>",
                    "block2": "I1<2>"
                }
            ],
            "const_name": "SymmBlock",
            "axis_dir": "V"
        }
    ]
}
```

stevenmburns, 2021-05-24:
There is a much simpler testcase in #717 . I'm going to close this issue a say that it is being addressed in that PR.

stevenmburns, 2021-05-28:
@854768750 @Lastdayends It would be great if one of you could work on this. I think a good name is "SameTemplate" but we can finalize that later once we get something implemented.

854768750, 2021-05-28:
sure

854768750, 2021-05-29:
After running `pytest -- tests/pnr/test_entrypoint2.py` I got this but I cannot find "row_entrypoint2" in $ALIGN_WORK_DIR.
[image]

854768750, 2021-05-29:
I am using the master branch

stevenmburns, 2021-05-29:
You sure. I set ALIGN_WORK_DIR to $ALIGN_HOME/work and I get that.

[p1]<https://user-images.githubusercontent.com/28666144/120052152-7cfa5d80-bfe9-11eb-91a0-f7c933d1d352.PNG>

stevenmburns, 2021-05-29:
Here is what I get:
```
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ rm -rf $ALIGN_WORK_DIR/{row_entrypoint2,matrix_entrypoint2}
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ ls $ALIGN_WORK_DIR/{row_entrypoint2,matrix_entrypoint2}
ls: cannot access '/home/smburns/DARPA/ALIGN-public/work/row_entrypoint2': No such file or directory
ls: cannot access '/home/smburns/DARPA/ALIGN-public/work/matrix_entrypoint2': No such file or directory
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ pytest -- tests/pnr/test_entrypoint2.py
=================================================== test session starts ====================================================
platform linux -- Python 3.8.0, pytest-6.2.3, py-1.10.0, pluggy-0.13.1
rootdir: /home/smburns/DARPA/ALIGN-public, configfile: pytest.ini
plugins: cov-2.11.1, forked-1.3.0, timeout-1.4.2, dash-1.20.0, xdist-2.2.1, cpp-1.5.0
collected 2 items                                                                                                          

tests/pnr/test_entrypoint2.py ..                                                                                     [100%]

------------------------------ generated xml file: /home/smburns/DARPA/ALIGN-public/junit.xml ------------------------------
==================================================== 2 passed in 1.36s =====================================================
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ ls $ALIGN_WORK_DIR/{row_entrypoint2,matrix_entrypoint2}
/home/smburns/DARPA/ALIGN-public/work/matrix_entrypoint2:
1_topology  2_primitives  3_pnr

/home/smburns/DARPA/ALIGN-public/work/row_entrypoint2:
1_topology  2_primitives  3_pnr

```

854768750, 2021-05-31:
Do I need to change anything in `schematic2layout.py inputs/ -s row --flow_start 3_pnr -n 40 --gui`？
I got ```[json.exception.type_error.302] type must be number, but is null```
It is probably the path is not correct, although the folder is not empty.

stevenmburns, 2021-05-31:
I get no JSON reading errors when I run it. Do you know what JSON file it is complaining about?
```
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/work/row_entrypoint2$ schematic2layout.py inputs/ -s row --flow_start 3_pnr -n 40 --gui
align.main INFO : Running flow steps ['3_pnr:prep', '3_pnr:place', '3_pnr:route', '3_pnr:check']
align.pnr.main INFO : Running Place & Route for row top_down ['3_pnr:prep', '3_pnr:place', '3_pnr:route', '3_pnr:check']
align.pnr.write_constraint INFO : found a SameTemplate: {'const_name': 'SameTemplate', 'blocks': ['u0', 'u1', 'u2']}
align.pnr.toplevel INFO : Starting bottom-up placement on row 0
PnR.placer.SeqPair.SetEnumerate INFO : Enumerated search
align.pnr.toplevel INFO : Working on row_0: {'width': 1.28, 'height': 1.68, 'hpwl': 1.616, 'cost': 23.440527878463055, 'constraint_penalty': 0.0, 'area_norm': 0.10666666666666667, 'hpwl_norm': 0.022357723577235773}
align.pnr.toplevel INFO : Working on row_1: {'width': 1.68, 'height': 1.68, 'hpwl': 2.416, 'cost': 23.91725600659271, 'constraint_penalty': 0.0, 'area_norm': 0.14, 'hpwl_norm': 0.027439024390243903}
align.pnr.toplevel INFO : Working on row_2: {'width': 2.48, 'height': 0.84, 'hpwl': 4.016, 'cost': 24.05540634507353, 'constraint_penalty': 0.0, 'area_norm': 0.10333333333333333, 'hpwl_norm': 0.042682926829268296}
align.pnr.toplevel INFO : Working on row_3: {'width': 1.68, 'height': 1.68, 'hpwl': 2.92, 'cost': 24.300248258848814, 'constraint_penalty': 0.0, 'area_norm': 0.14, 'hpwl_norm': 0.04024390243902439}
align.pnr.toplevel INFO : Working on row_4: {'width': 2.48, 'height': 0.84, 'hpwl': 4.52, 'cost': 24.317770609541018, 'constraint_penalty': 0.0, 'area_norm': 0.10333333333333333, 'hpwl_norm': 0.05548780487804878}
align.pnr.toplevel INFO : Working on row_5: {'width': 1.28, 'height': 1.68, 'hpwl': 2.96, 'cost': 24.367668626361276, 'constraint_penalty': 0.0, 'area_norm': 0.10666666666666667, 'hpwl_norm': 0.056504065040650406}
align.pnr.toplevel INFO : Working on row_6: {'width': 1.68, 'height': 1.68, 'hpwl': 2.92, 'cost': 24.419152999614397, 'constraint_penalty': 0.0, 'area_norm': 0.14, 'hpwl_norm': 0.04532520325203252}
align.pnr.toplevel INFO : Working on row_7: {'width': 2.08, 'height': 1.68, 'hpwl': 3.216, 'cost': 24.445911153530666, 'constraint_penalty': 0.0, 'area_norm': 0.17333333333333334, 'hpwl_norm': 0.037601626016260166}
align.pnr.toplevel INFO : Working on row_8: {'width': 1.68, 'height': 1.68, 'hpwl': 3.256, 'cost': 24.49262015149627, 'constraint_penalty': 0.0, 'area_norm': 0.14, 'hpwl_norm': 0.04878048780487805}
align.pnr.toplevel INFO : Working on row_9: {'width': 2.48, 'height': 0.84, 'hpwl': 5.024, 'cost': 24.525409974319263, 'constraint_penalty': 0.0, 'area_norm': 0.10333333333333333, 'hpwl_norm': 0.06829268292682927}
align.pnr.toplevel INFO : Working on row_10: {'width': 3.28, 'height': 0.84, 'hpwl': 5.616, 'cost': 24.54856530759075, 'constraint_penalty': 0.0, 'area_norm': 0.13666666666666666, 'hpwl_norm': 0.052845528455284556}
align.pnr.toplevel INFO : Working on row_11: {'width': 3.28, 'height': 0.84, 'hpwl': 5.7, 'cost': 24.58815577486176, 'constraint_penalty': 0.0, 'area_norm': 0.13666666666666666, 'hpwl_norm': 0.054979674796747965}
align.pnr.toplevel INFO : Working on row_12: {'width': 1.68, 'height': 1.68, 'hpwl': 3.256, 'cost': 24.591711054140504, 'constraint_penalty': 0.0, 'area_norm': 0.14, 'hpwl_norm': 0.05386178861788618}
align.pnr.toplevel INFO : Working on row_13: {'width': 2.48, 'height': 1.68, 'hpwl': 4.016, 'cost': 24.62180181999433, 'constraint_penalty': 0.0, 'area_norm': 0.20666666666666667, 'hpwl_norm': 0.037601626016260166}
align.pnr.toplevel INFO : Working on row_14: {'width': 2.48, 'height': 1.68, 'hpwl': 4.1, 'cost': 24.677006374341424, 'constraint_penalty': 0.0, 'area_norm': 0.20666666666666667, 'hpwl_norm': 0.039735772357723576}
align.pnr.toplevel INFO : Working on row_15: {'width': 4.88, 'height': 0.42, 'hpwl': 8.816, 'cost': 24.708195453182633, 'constraint_penalty': 0.0, 'area_norm': 0.10166666666666667, 'hpwl_norm': 0.08333333333333333}
align.pnr.toplevel INFO : Working on row_16: {'width': 1.68, 'height': 1.68, 'hpwl': 3.76, 'cost': 24.72571403366365, 'constraint_penalty': 0.0, 'area_norm': 0.14, 'hpwl_norm': 0.061585365853658536}
align.pnr.toplevel INFO : Working on row_17: {'width': 3.28, 'height': 0.84, 'hpwl': 6.036, 'cost': 24.73248814575168, 'constraint_penalty': 0.0, 'area_norm': 0.13666666666666666, 'hpwl_norm': 0.06351626016260163}
align.pnr.toplevel INFO : Working on row_18: {'width': 4.88, 'height': 0.42, 'hpwl': 8.9, 'cost': 24.733482772897283, 'constraint_penalty': 0.0, 'area_norm': 0.10166666666666667, 'hpwl_norm': 0.08546747967479675}
align.pnr.toplevel INFO : Working on row_19: {'width': 2.08, 'height': 1.68, 'hpwl': 3.72, 'cost': 24.73898407461732, 'constraint_penalty': 0.0, 'area_norm': 0.17333333333333334, 'hpwl_norm': 0.05040650406504065}
align.pnr.toplevel INFO : Working on row_20: {'width': 3.28, 'height': 0.84, 'hpwl': 5.7, 'cost': 24.757765952935948, 'constraint_penalty': 0.0, 'area_norm': 0.13666666666666666, 'hpwl_norm': 0.06514227642276423}
align.pnr.toplevel INFO : Working on row_21: {'width': 4.88, 'height': 0.42, 'hpwl': 8.984, 'cost': 24.758146383588027, 'constraint_penalty': 0.0, 'area_norm': 0.10166666666666667, 'hpwl_norm': 0.08760162601626016}
[... cut ...]
align.pnr.main ERROR : 2 LVS / DRC errors found !!!
align.pnr.main INFO : OUTPUT error file at /home/smburns/DARPA/ALIGN-public/work/row_entrypoint2/3_pnr/row_0.errors
align.pnr.main INFO : OUTPUT gds.json /home/smburns/DARPA/ALIGN-public/work/row_entrypoint2/3_pnr/row_0.python.gds.json
Use KLayout to visualize the generated GDS: /home/smburns/DARPA/ALIGN-public/work/row_entrypoint2/row_0.gds
Use KLayout to visualize the python generated GDS: /home/smburns/DARPA/ALIGN-public/work/row_entrypoint2/row_0.python.gds
```

854768750, 2021-06-01:
I added some lines in #724 for this new feature.

stevenmburns, 2021-06-02:
Closed due to #724

closed 2021-06-02 by stevenmburns