ALIGN-analoglayout/ALIGN-public #718: Extra space between subbocks for no apparent reason.

stevenmburns, 2021-05-24:
[image]

There seems to be extra space between subblocks in placer generated placements. The subblock dimensions are on grid.

To duplicate try (assuming #717 has been merged, or in #717 if not):
```
pytest -- tests/pnr/test_entrypoint2.py 
export ALIGN_WORK_DIR=$ALIGN_HOME/work
cd $ALIGN_WORK_DIR/row_entrypoint2
schematic2layout.py inputs/ -s row --flow_start 3_pnr -n 40 --gui 
```
All placements seem to have extra space in the X direction between the first and the other blocks.

stevenmburns, 2021-05-25:
I've merged #717

stevenmburns, 2021-05-26:
@854768750 @Lastdayends Any thoughts on how to fix this?

854768750, 2021-06-02:
I added some lines in #727 . It seems correct now.

parijatm, 2022-01-21:
Addressed by above mentioned PRs

closed 2022-01-21 by parijatm