ALIGN-analoglayout/ALIGN-public #723: Enable flipping cells in both X and Y.

stevenmburns, 2021-05-28:
Is there a way to get the placer to consider flipping a cell across the Y-axis (change the x coordinate) Same thing for mirror across the X-axis. I have some synthetic testcases that would be a lot better in HPWL if we allowed flipping (or I know how to turn in on.)

@Lastdayends @854768750 @srini229 Any thoughts on how to do this? Can we at least consider doing it in the enumerative placer if the number of cells is very small

srini229, 2021-05-28:
@stevenmburns We can add this step as one more variant generator in the sequence pair modify routine. I can try it out with the enumerative placer and keep you posted.

stevenmburns, 2021-05-29:
@srini229  That would be great.

srini229, 2021-06-01:
@stevenmburns I added the flip enumerator to the existing enumerated placer; You can find the change in the branch origin/flip_x_y_enum_placer. I verified it on a cooked up testcase with chain of inverters. Please let me know if you don't see expected behaviour with this change.

stevenmburns, 2021-06-02:
Closed due to #725

closed 2021-06-02 by stevenmburns