ALIGN-analoglayout/ALIGN-public #730: nvariants and effort

soneryaldiz, 2021-06-09:
```python
parser.add_argument("-n", "--nvariants", type=int, default=1,
                            help='Number of layout candidates to (attempt to) generate')
parser.add_argument("-e", "--effort", type=int, default=0,
                            help='Amount of effort to dedicate to alternate layouts')
```

The commands below generate the exact same placement:
`schematic2layout.py $ALIGN_HOME/align/pdk/finfet/examples/comparator -p $ALIGN_HOME/align/pdk/finfet`
`schematic2layout.py $ALIGN_HOME/align/pdk/finfet/examples/comparator -p $ALIGN_HOME/align/pdk/finfet -e 2`
`schematic2layout.py $ALIGN_HOME/align/pdk/finfet/examples/comparator -p $ALIGN_HOME/align/pdk/finfet -e 4`

The first (optimal) solution found by the commands below are different:
`schematic2layout.py $ALIGN_HOME/align/pdk/finfet/examples/comparator -p $ALIGN_HOME/align/pdk/finfet`
`schematic2layout.py $ALIGN_HOME/align/pdk/finfet/examples/comparator -p $ALIGN_HOME/align/pdk/finfet -n 2`
`schematic2layout.py $ALIGN_HOME/align/pdk/finfet/examples/comparator -p $ALIGN_HOME/align/pdk/finfet -n 4`

1. Could you please explain the differences between --nvariants and --effort?
3. Why are the optimal solutions found by varying n are different?
4. Why does increasing effort generate the same layout while nvariants find alternative solutions?

srini229, 2021-06-09:
@soneryaldiz **nvariants** sets the number of variants at all hierarchies. Choice of most optimal solution at a lower hierarchy may not be optimal higher up. I think that's what is happening here. Ideally we should always choose multi variants for intermediate and leaf hierarchies independent of how many variants the user asks and return the **nvariants** only for the top.

854768750, 2021-06-09:
"--effort" determines the multiplier of the number of simulated annealing iterations.
#iterations = #initial iterations * ((effort == 0) ? 1. : ((effort == 1) ? 4. : 8.))

854768750, 2021-06-09:
There are 3 choices of effort (0, 1, else). Effort==2 and effort==3 generate the same result.
Effort==0 and effort==1 may generated different results, but may also generate the same because increasing the number of iterations of simulated annealing does not always decrease the cost value.

soneryaldiz, 2021-06-09:
@srini229 , I agree with you, the placer should always consider variants for subhierarchy. I have several examples where the generated layout is very suboptimal without _nvariants_. I will present few this Thursday.

@854768750 , what does _nvariants_ control exactly? Does it control how many variants to generate per each level of hierarchy? What is a recommended typical value?

854768750, 2021-06-09:
It controls how many variants to generate per each level of hierarchy **at most**. For example, for a 2 block hierarchy, if n ==100, the number of generated variants may be less than 100.

soneryaldiz, 2021-06-09:
What does a variant mean for a primitive? Does it mean different aspect ratios and flips?

What does a variant mean for a hierarchy above primitive?

854768750, 2021-06-09:
For a primitive, a variant means different concrete templates.
For a hierarchy above primitive, it means different placement results.

854768750, 2021-06-09:
When varying n, the number of intermediate variants is different. Then the best top level results is also different when varying n.

854768750, 2021-06-09:
Returning the nvariants only for the top seems more reasonable.

soneryaldiz, 2021-06-09:
My take-away from this thread is to always specify -n. 

The definition for primitives is clear and the number variants is capped at 12 as of Today in align/main.py.

For intermediate hierarchy, what does different mean? Just randomly generated n different placements that satisfy constraints if given? Is there any cost function involved in selecting top n for intermediate hierarchy?

854768750, 2021-06-09:
Intermediate hierarchy uses the same cost function as the top.