ALIGN-analoglayout/ALIGN-public #731: Half Perimeter Wire Lenght (HPWL) Example

soneryaldiz, 2021-06-09:
How is HPWL formulated within placer for the example below? Could you please express in terms of wire segments:
<_block name_>.<_wire number_>.<_coordinate llx, lly, urx, ury, mx, my_> where A.1.mx = ( A.1.llx + A.1.urx ) / 2

[image]

854768750, 2021-06-09:
(C.1.mx-A.1.mx)+(A.1.my-B.3.my)

stevenmburns, 2021-06-09:
@854768750 @Lastdayends What is the rationale for using the midpoints instead of the full extent of the wire? An alternative would be to use:
C.1.urx-A.1.llx + A.1.ury-B.2.lly
Is that harder to implement? (I think you need to do about the same work if your block has hierarchy.) It seems to me that it would give a better estimate for the capacitance of the net than using the center points.

854768750, 2021-06-09:
Is the HPWL also calculated anywhere else? For example, in the checker?
I just want to keep the HPWL formulation the same.
If nowhere else calculates HPWL, C.1.urx-A.1.llx + A.1.ury-B.2.lly is a better estimate.

soneryaldiz, 2021-06-09:
I would suggest another alternative which excludes the existing tracks within the primitive as they are unavoidable anyway and bias the solution. How about something like below which is the minimum reasonable box to connect these terminals?
`(C.1.llx - A.1.urx) + (A.1.lly - C.1.ury)`
There might be a way to generalize this.

stevenmburns, 2021-06-10:
@soneryaldiz I don't know how to compute the smaller/reasonable box. I also don't know why we shouldn't include the wire that are already there as part of the capacitance estimate.

@854768750 We haven't started computing the HPWL in the checker yet. I'm computing it for the GUI and I am currently using the scheme I proposed above. I think everything should shift to that metric and we can compare the values to make sure we are computing them consistently. You can look in align/pnr/hpwl.py for two ways to compute the metric. One interesting thing is that internal layers of hierarchy can contribute to the HPWL because of wires that do not go higher in the hierarchy (and we should count this because different implementation of a sub-hierarchy can have different amounts of "internal" HPWL, and because we don't want this metric  to change when we add or delete a hierarchy to help with writing constraints.)
I am also including the global nets as well. Perhaps we don't want to include these. I can add a switch to select whether or not to include global nets in this metric.

@parijatm You might be interested in this discussion too.

854768750, 2021-06-10:
I added something to calculate "C.1.urx-A.1.llx + A.1.ury-B.2.lly" in https://github.com/ALIGN-analoglayout/ALIGN-public/pull/733/commits/63fab088fab5b5a6508df01951be3c6073636058#diff-eeece304a5487cdaf0321dbf7cf00c10d3932afaacb1f761718b075d95821d2cR258

soneryaldiz, 2021-06-14:
@stevenmburns , if total capacitance is the key concern, HPWL is not a good estimator for mesh-style routing. Please see the example which shows two different aspect ratios of an hypothetical primitive. For both scenarios, HPWL underestimates the total capacitance. That is why I was thinking of focusing only on the capacitance of additional wires the router will add.
[image]

Regardless of which formula we use, the hierarchy should not impact the calculation as you pointed out and this should be fixed. Is there an issue tracking this?

As there is a grid for global nets, those should be excluded from the calculation.

stevenmburns, 2021-06-14:
@soneryaldiz
@854768750 and I are working on correcting the HPWL calculation in PR #733 The next step is correct hierarchical calculation.

Just because the HPWL is only a lower bound on the capacitance doesn't mean it is not a good metric for placement.  We can keep thinking about the right metric, but let's define HPWL and compute it correctly before we move on to something else. Right now, we are not computing HPWL correctly and that is leading to sub-optimal placements.

soneryaldiz, 2021-06-14:
> @soneryaldiz
> @854768750 and I are working on correcting the HPWL calculation in PR #733 The next step is correct hierarchical calculation.
>
> Just because the HPWL is only a lower bound on the [...]

Sounds good. I am closing this issue as the question has been answered.

closed 2021-06-14 by soneryaldiz