ALIGN-analoglayout/ALIGN-public #737: setup issue during pytest

PDN-BeGAN, 2021-06-12:
ImportError while loading conftest '/ALIGN-public/conftest.py'.
conftest.py:2: in <module>
    from align.utils.logging import reconfigure_loglevels
align/__init__.py:3: in <module>
    from .utils import logging
align/utils/logging.py:1: in <module>
    import logging.config
E   ImportError: No module named config

stevenmburns, 2021-06-14:
What version of python are you using?
What steps are you using to install the flow?

Following the steps in README.md work for most of us. Did the pip install work correctly?

kkunal1408, 2021-06-14:
Pip install worked fine. I am using python 3.8.

parijatm, 2021-06-14:
Unable to reproduce locally. Definitely system dependent. 

Potential fix based on error signature in #738. Waiting on @kkunal1408 to test.

stevenmburns, 2021-06-15:
I tried to rebuild in a new virtual environment in several different ways.
The development builds work.
The user build doesn't.

@parijatm Maybe we should change the README.md so that we don't suggest that as a way to do build it.

Fails:
```
pip install -v .
```

Works:
```
pip install -e .[test]
```

Works:
```
pip install setuptools wheel pybind11 scikit-build cmake ninja
pip install -v -e .[test] --no-build-isolation
```

Works (what I recommend if you are a python developer)
```
pip install setuptools wheel pybind11 scikit-build cmake ninja
pip install -v -e .[test] --no-build-isolation
pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=Release'
```

```

PDN-BeGAN, 2021-06-15:
It worked on a different system. In this system, I figured out starting from scratch failed during installation at this step.
   CMake Error at CMakeLists.txt:8 (project):
      Running

       '/tmp/pip-build-env-d6e74y9s/overlay/bin/ninja' '--version'

      failed with:

       No such file or directory
I think it's related to my system and some local system related issues. I will check more and update.

kkunal1408, 2021-06-15:
It worked on a different system. In this system, I figured out starting from scratch failed during installation at this step.
   CMake Error at CMakeLists.txt:8 (project):
      Running

       '/tmp/pip-build-env-d6e74y9s/overlay/bin/ninja' '--version'

      failed with:

       No such file or directory
I think it's related to my system and some local system-related issues. I will check more and update.

parijatm, 2021-06-15:
> Fails:
> 
> ```
> pip install -v .
> ```
> 

What is the failure you are getting @stevenmburns ? Is it `cannot import name 'PnR' from partially initialized module 'align'`? If so, align has been installed correctly. We are just invoking it from the wrong directory. CDing to a different directory and running schematic2layout will work just fine. 

We can still remove it if it is causing confusion but the long term solution would be to replace the 'align' directory with 'src' to avoid the namespace confusion  in the first place (Google 'tox src layout' for details). I have currently hacked around this by deleting the 'align' directory when testing it in user mode in our regressions.

stevenmburns, 2021-06-15:
The pip install itself fails. Here is the end of the error log:
```
...
  Configuring Project
    Working directory:
      /tmp/pip-req-build-xf5ew3uy/_skbuild/linux-x86_64-3.8/cmake-build
    Command:
      cmake /tmp/pip-req-build-xf5ew3uy -G Ninja -DCMAKE_INSTALL_PREFIX:PATH=/tmp/pip-req-build-xf5ew3uy/_skbuild/linux-x86_64-3.8/cmake-install -DPYTHON_EXECUTABLE:FILEPATH=/home/smburns/DARPA/ALIGN-public/.venv2/bin/python -DPYTHON_VERSION_STRING:STRING=3.8.0 -DPYTHON_INCLUDE_DIR:PATH=/usr/include/python3.8 -DPYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython3.8.so -DSKBUILD:BOOL=TRUE -DCMAKE_MODULE_PATH:PATH=/tmp/pip-build-env-xevgcl07/overlay/lib/python3.8/site-packages/skbuild/resources/cmake -DALIGN_VERSION:string=0.9.8 -DCMAKE_BUILD_TYPE:STRING=Release

  Traceback (most recent call last):

  An error occurred while configuring with CMake.
    Command:
      cmake /tmp/pip-req-build-xf5ew3uy -G Ninja -DCMAKE_INSTALL_PREFIX:PATH=/tmp/pip-req-build-xf5ew3uy/_skbuild/linux-x86_64-3.8/cmake-install -DPYTHON_EXECUTABLE:FILEPATH=/home/smburns/DARPA/ALIGN-public/.venv2/bin/python -DPYTHON_VERSION_STRING:STRING=3.8.0 -DPYTHON_INCLUDE_DIR:PATH=/usr/include/python3.8 -DPYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython3.8.so -DSKBUILD:BOOL=TRUE -DCMAKE_MODULE_PATH:PATH=/tmp/pip-build-env-xevgcl07/overlay/lib/python3.8/site-packages/skbuild/resources/cmake -DALIGN_VERSION:string=0.9.8 -DCMAKE_BUILD_TYPE:STRING=Release
    Source directory:
      /tmp/pip-req-build-xf5ew3uy
    Working directory:
      /tmp/pip-req-build-xf5ew3uy/_skbuild/linux-x86_64-3.8/cmake-build
  Please see CMake's output for more information.
  Building wheel for align (PEP 517) ... error
  ERROR: Failed building wheel for align
Failed to build align
ERROR: Could not build wheels for align which use PEP 517 and cannot be installed directly
Exception information:
Traceback (most recent call last):
  File "/home/smburns/DARPA/ALIGN-public/.venv2/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 180, in _main
    status = self.run(options, args)
  File "/home/smburns/DARPA/ALIGN-public/.venv2/lib/python3.8/site-packages/pip/_internal/cli/req_command.py", line 204, in wrapper
    return func(self, options, args)
  File "/home/smburns/DARPA/ALIGN-public/.venv2/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 360, in run
    raise InstallationError(
pip._internal.exceptions.InstallationError: Could not build wheels for align which use PEP 517 and cannot be installed directly
Removed build tracker: '/tmp/pip-req-tracker-ff6or4v4'

```

parijatm, 2021-06-15:
> The pip install itself fails. Here is the end of the error log:

I think its a cmake issue (you will need to scroll a bit further up to be sure). Removing the _skbuild directory should resolve it. 

The error is encountered anytime you switch from one installation target to another. So going from `pip install` to `pip install .` will likely also result in the same error.

Side note: Looks like merging #738 closed this issue automatically. Reopening as there are a few opens to resolve.

stevenmburns, 2021-06-15:
Yes, removing _skbuild makes the build work. (Maybe we need to add this in the README.md.)
Now, I can't get the tests to run because it can't find the module 'colorlog' although it is install and can be imported from a fresh python REPL.

There are weird things going on.
```
(.venv2) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ pytest
ImportError while loading conftest '/home/smburns/DARPA/ALIGN-public/conftest.py'.
conftest.py:2: in <module>
    from align.utils import logmanager
align/__init__.py:3: in <module>
    from .utils import logmanager
align/utils/logmanager.py:4: in <module>
    import colorlog
E   ModuleNotFoundError: No module named 'colorlog'
(.venv2) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ python
Python 3.8.0 (default, Feb 25 2021, 22:10:10) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import colorlog
>>> exit()
```

stevenmburns, 2021-06-15:
Here is a smoking gun:
```
(.venv2) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ pytest
ImportError while loading conftest '/home/smburns/DARPA/ALIGN-public/conftest.py'.
conftest.py:2: in <module>
    from align.utils import logmanager
align/__init__.py:3: in <module>
    from .utils import logmanager
align/utils/logmanager.py:4: in <module>
    import colorlog
E   ModuleNotFoundError: No module named 'colorlog'
(.venv2) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ python -m pytest
====================================================== test session starts =======================================================
platform linux -- Python 3.8.0, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
rootdir: /home/smburns/DARPA/ALIGN-public, configfile: pytest.ini, testpaths: tests
plugins: cov-2.12.1, forked-1.3.0, timeout-1.4.2, cpp-2.0.0, dash-1.20.0, xdist-2.2.1
collected 2360 items                                                                                                             

tests/test_build_steps.py .                                                                                                [  0%]
tests/test_cmdline.py ..^C
```

?

parijatm, 2021-06-15:
I am a bit flummoxed !! 😄 

Looks like the `pytest` command is not pointing to the .venv2/bin/pytest. Can you try a `which pytest` to see if that's the case?

If that is the case, its possible pytest wasn't installed in .venv2 and its instead using the python2 pytest which doesn't have colorlog installed.

PR #738 (2021-06-15): Rename align.utils.logging
Unable to reproduce #737 on my end but refactoring filenames to avoid global module names is a good best practice anyway.

@kkunal1408 can you please test this branch on your setup and see if it resolves #737?