ALIGN-analoglayout/ALIGN-public #741: build-type to be Release (-O3)

AmrMohamedRashed, 2021-06-17:
I am trying to run the following command as I want to build ALIGN as a python developer 
Command: pip install -v -e .[test] --no-build-isolation
Error sample: https://drive.google.com/file/d/11DalBYvgC8yJtln525UybXfsKsclpVCU/view?usp=sharing

and also the following command gives an error similar to the previous one
Command: pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=Release'
Error sample: https://drive.google.com/file/d/10EP0MuZBOCzpL7X8cW6djnlfbdHCDtU3/view?usp=sharing

Finally I tried the following command with the following error:
Command: python -m cProfile -o stats $ALIGN_HOME/bin/schematic2layout.py $ALIGN_HOME/examples/sc_dc_dc_converter
Error: FileNotFoundError: [Errno 2] No such file or directory: '/bin/schematic2layout.py'
While the file "schematic2layout.py" exists in bin folder.

stevenmburns, 2021-06-17:
@parijatm Maybe you can have a look here. I can't tell what is happening.

@AmrMohamedRashed For the last failure, you need to set the environment variable `ALIGN_HOME` to be the directory of the git checkout. It will likely fail because of install issues, but that is the reason for the particular error above.

parijatm, 2021-06-18:
Could you please share the full build log @AmrMohamedRashed ? Compilation issues (if any) show up prior to the snippet you posted.

AmrMohamedRashed, 2021-06-18:
@parijatm here is the command and the full log of the command: 
Command: pip install -v -e .[test] --no-build-isolation
Error: https://drive.google.com/file/d/11mxTq9PB_Lof1bwkW-dHZcLRVO5RfPAL/view?usp=sharing

Command: pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=Release'
Error: https://drive.google.com/file/d/1Gu8qUdkiH_rDMwOo796Q_Oqex-aiOisy/view?usp=sharing

parijatm, 2021-06-18:
@AmrMohamedRashed your cmake seems to be having issues finding ninja.

Can you please activate your virtualenv and try `ninja --version`? 

If it says ninja not found, please run the following command once more and make sure it passes without any issue:
```console
$ pip install setuptools wheel pybind11 scikit-build cmake ninja
```

parijatm, 2021-06-18:
I just noticed the following:

> I am trying to run the following command as I want to build ALIGN as a python developer

Can you just try running `pip install -e -v .[test]` and see if it works out of the box? Please remove the local _skbuild directory before doing so. 

It is possible we may want to simplify our build flow and make the default build type to Release. We could enable Debug mode only for C++ developers @stevenmburns.

AmrMohamedRashed, 2021-06-18:
@parijatm here is the output of ninja --version
1.10.0.git.kitware.jobserver-1
but running this command "pip install -v -e .[test] --no-build-isolation" after this command "pip install setuptools wheel pybind11 scikit-build cmake ninja" still gives the same error as previous.
------------------------------------------------------------------------------
Running "pip install -e -v .[test]" after deleting local folder _skbuild gives the following error:
ERROR: -v is not a valid editable requirement. It should either be a path to a local project or a VCS URL (beginning with bzr+http, bzr+https, bzr+ssh, bzr+sftp, bzr+ftp, bzr+lp, bzr+file, git+http, git+https, git+ssh, git+git, git+file, hg+file, hg+http, hg+https, hg+ssh, hg+static-http, svn+ssh, svn+http, svn+https, svn+svn, svn+file).

parijatm, 2021-06-18:
> @parijatm here is the output of ninja --version
> 1.10.0.git.kitware.jobserver-1
> but running this command "pip install -v -e .[test] --no-build-isolation" after this command "pip install setuptools [...]

Sorry it should be `pip install -v -e .[test]`

AmrMohamedRashed, 2021-06-21:
@parijatm still same error.

closed 2021-06-21 by AmrMohamedRashed

stevenmburns, 2021-06-21:
@AmrMohamedRashed Did it work for you in the end?

AmrMohamedRashed, 2021-06-21:
@stevenmburns Yes thank you and thank you @parijatm

parijatm, 2021-06-21:
I met with Amr. It works out of the box with `pip install -v -e .[test]`. We did not debug the issues causing the other installation modes to fail. To simplify instructions, I think we should switch the default CMAKE built style to Release and have the C++ developers build with Debug mode. What do you think @stevenmburns ?

parijatm, 2021-06-21:
Small note in case something of the sort crops up again (may not be pertinent). The reported issue was faced on Anaconda.