ALIGN-analoglayout/ALIGN-public #780: DRC in adder circuit near capacitor pins

kkunal1408, 2021-08-12:
[image]
I found a problem. The open happens when routing cap pins. And the cap pins are out of the block boundary.
[image]
And these two pins should be connected. But their distance is only one M2 pitch

_Originally posted by @854768750 in https://github.com/ALIGN-analoglayout/ALIGN-public/issues/615#issuecomment-893825885_

kkunal1408, 2021-08-12:
It looks like it might have trouble routing between the pins even if they are spaced farther apart. The top blue pin (on the resistor) has to connect to the blue pin below connected to the capacitor. There are likely obstructions there because of the m3 in the capacitor. It would have to do wrong way routing to make the connection and avoid the m3 obstructions. Do you have a visualization of what the global router does in this case?

[image]

_Originally posted by @stevenmburns in https://github.com/ALIGN-analoglayout/ALIGN-public/issues/615#issuecomment-893878291_

kkunal1408, 2021-08-12:
Moving this issue seperately.

arvuce22, 2021-08-13:
Updated the boundary for the cap in: https://github.com/ALIGN-analoglayout/ALIGN-public/tree/feature/brave_new_world

[image]

arvuce22, 2021-11-12:
Cap pin issue is resolved in #830, now adder example does not have opens. However, there are issues in SCF test cases we are trying to resolve those.

arvuce22, 2021-12-05:
#830 addressed this issue.

closed 2021-12-05 by arvuce22