ALIGN-analoglayout/ALIGN-public #781: Tests don't all pass when compiled with Release mode and RelWIthDebInfo mode

stevenmburns, 2021-08-12:
@srini229 @854768750 @Lastdayends @kkunal1408 Even on master, our tests are not passing if we compile using one of the optimized build types. We need these to work.

```console
$ pip install setuptools wheel pybind11 scikit-build cmake ninja
$ pip install -v -e .[test] --no-build-isolation
$ pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=Release'
```
or
```console
$ pip install setuptools wheel pybind11 scikit-build cmake ninja
$ pip install -v -e .[test] --no-build-isolation
$ pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=RelWithDebInfo'
```
One of the tests that is failing can be reproduced using this command line (make sure ALIGN_HOME and ALIGN_WORK_DIR are set.)

```
cd $ALIGN_WORK_DIR
mkdir high_speed_comparator
cd high_speed_comparator
schematic2layout.py $ALIGN_HOME/examples/high_speed_comparator/ --skipGDSschematic2layout.py ../../examples/high_speed_comparator/ --skipGDS
```

854768750, 2021-08-13:
I used `--build-type=Release` and run `schematic2layout.py $ALIGN_HOME/examples/high_speed_comparator/ --skipGDS` but there was no error.

854768750, 2021-08-13:
> Another test is failing. Duplicate using:
> 
> ```
> python -m pytest -s -vv -k 'test_order_abut'
> ```

This is also correct on my computer. 
I used 
`
$ pip install setuptools wheel pybind11 scikit-build cmake ninja
$ pip install -v -e .[test] --no-build-isolation
$ pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=Release'
`
and run the above two command lines but got no error.

kkunal1408, 2021-08-27:
@stevenmburns is this issue resolved, or are we still facing these issues?

stevenmburns, 2021-08-28:
This was resolved by #783

closed 2021-08-28 by stevenmburns