ALIGN-analoglayout/ALIGN-public #790: Code quality button is not working on master.

kkunal1408, 2021-08-28:


stevenmburns, 2021-08-30:
@kkunal1408 Code quality is broken but coverage is still available. Click on the green "PASSED" badge at the beginning of a rendered README.md. Find "build-test-coverage" in the list of CI jobs. Click on that.
[image]
Click on artifacts:
[image]
Then click on `coverage-reports/python/index.html` and scroll down to the compiler code:

[image]

Looks like `read_netlist.py` and `merge_nodes.py` aren't used in any tests anymore. `preprocess.py` is only enabling the logger. Probably a lot you can get rid of.

kkunal1408, 2021-12-10:
the workaround is good. There is separate PR for codacy.

closed 2021-12-10 by kkunal1408