ALIGN-analoglayout/ALIGN-public #795: Possible implementation of Casefolding String

stevenmburns, 2021-09-11:
@parijatm Here might be a way to implement casefolding strings.
It seems to assert appropriately in both comparisons with a standard string.

```
import pytest
from collections.abc import Sequence

class SpiceStr(Sequence):
    def __init__(self, s):
        self.s = s

    def __eq__(self, other):
        assert type(other) == SpiceStr
        return self.s.casefold() == other.s.casefold()

    def __hash__(self):
        return hash(self.s.casefold())

    def __str__(self):
        return s

    def __contains__(self, t):
        return self.s.casefold().__contains__( t.casefold())

    def __getitem__(self, i):
        return self.s[i]

    def __len__(self):
        return len(self.s)

def test_A():
    assert SpiceStr("Steve") == SpiceStr("steve")
    assert SpiceStr("Steve") != SpiceStr("burns")
    with pytest.raises(AssertionError):
        SpiceStr("Steve") == "steve"

    with pytest.raises(AssertionError):
        "steve" != SpiceStr("Steve")

    assert "Ste" in SpiceStr("steve")
    assert not "Burns" in SpiceStr("steve")

    # Get these because we are inheriting from collections.abc.Sequence

    assert SpiceStr("Steve")[0] == 'S'
    assert len(SpiceStr("Steve")) == 5

    assert list(reversed(SpiceStr("Steve"))) == list("evetS")

    assert list("Steve") == [c for c in SpiceStr("Steve")]
```

stevenmburns, 2021-09-11:
@parijatm Looks like you can make comparisons with regular strings work as well (you don't need to assert---maybe we should anyway):
```
import pytest
from collections.abc import Sequence

class SpiceStr(Sequence):
    def __init__(self, s):
        self.s = s

    def __eq__(self, other):
        t = other if type(other) == SpiceStr else SpiceStr(other)
        assert type(t) == SpiceStr
        return self.s.casefold() == t.s.casefold()

    def __hash__(self):
        return hash(self.s.casefold())

    def __str__(self):
        return self.s

    def __contains__(self, t):
        tt = str(t) if type(t) == SpiceStr else t
        return self.s.casefold().__contains__( tt.casefold())

    def __getitem__(self, i):
        return self.s[i]

    def __len__(self):
        return len(self.s)

def test_A():
    assert SpiceStr("Steve") == SpiceStr("steve")
    assert SpiceStr("Steve") != SpiceStr("burns")

    assert SpiceStr("Steve") == "steve"

    assert "steve" == SpiceStr("Steve")

    assert "burns" != SpiceStr("Steve")

    assert "Ste" in SpiceStr("steve")
    assert not "Burns" in SpiceStr("steve")

    assert SpiceStr("Ste") in SpiceStr("steve")

    # Get these because we are inheriting from collections.abc.Sequence

    assert SpiceStr("Steve")[0] == 'S'
    assert len(SpiceStr("Steve")) == 5

    assert list(reversed(SpiceStr("Steve"))) == list("evetS")

    assert list("Steve") == [c for c in SpiceStr("Steve")]

def test_B():
    s = { SpiceStr("Steve"), SpiceStr("Burns")}
    assert SpiceStr("Steve") in s
    assert SpiceStr("steve") in s

    assert "steve" in s
    assert "foo" not in s
```

parijatm, 2021-09-13:
Looks really good Steve ! Having said that, the following implementation appears to satisfy all tests as well.

```python
class SpiceStr(str):

    def __eq__(self, other):
        return super().casefold() == other.casefold()

    def __hash__(self) -> int:
        return super().casefold().__hash__()

    def __contains__(self, other):
        return super().casefold().__contains__(other.casefold())
```

What was unexpected is that the following works for both implementations:
```python
assert SpiceStr("Steve") == "StEve"
assert "StEve" == SpiceStr("Steve")
```
Which basically means that `SpiceStr.__eq__()` is called irrespective of whether the `SpiceStr` object is on the left or right of the == expression. So my previous concerns were a non-issue.

stevenmburns, 2021-09-13:
Even better. I'd leave a comment in there (or these tests) so that future maintainers know that this works since it wasn't obvious to us. And should you use `str` or `collections.UserString` as the base class?

parijatm, 2021-09-13:
I am adding the tests directly to #794 as well. Can also leave a pointer to this issue somewhere in the code.

As far as str vs. collections.UserString goes, I stuck with str upon reading [this](https://docs.python.org/3/library/collections.html#collections.UserString):
> The class, UserString acts as a wrapper around string objects. The need for this class has been partially supplanted by the ability to subclass directly from str; however, this class can be easier to work with because the underlying string is accessible as an attribute.

I haven't seen the need to directly access the unmodified underlying string (yet). We can switch to UserString if we encounter said need.

srini229, 2021-09-13:
Relevant blob I found in SO. Rich compare is tried both ways for existence of overloading.

https://github.com/python/cpython/blob/v3.5.10/Objects/object.c#L649-L699