ALIGN-analoglayout/ALIGN-public #811: Mapping ALIGN to new PDK

AmrMohamedRashed, 2021-09-30:
I want to map the tool to work with tsmc65 PDK that I have on my PC. Is there anyway or script to generate "layers.json" for the new PDK? 
or is there any instruction/tutorial to map the tool to new PDK?

thank you in advance.

AmrMohamedRashed, 2021-10-05:
@arvuce22 looking forward to it

AmrMohamedRashed, 2021-11-28:
@arvuce22  is there any update regarding layer.json generation?

arvuce22, 2021-11-28:
@AmrMohamedRashed Yes, we have been working on it and will provide a document to generate "Layers.json" by next week.

AmrMohamedRashed, 2021-12-06:
@arvuce22 would generating "Layers.json" be enough for mapping ALIGN to new pdk?

AmrMohamedRashed, 2022-01-02:
@arvuce22 is there any update regarding mapping the tool to a new PDK?

arvuce22, 2022-01-05:
Can we have a meeting sometime this week?

arvuce22, 2022-01-05:
@AmrMohamedRashed Can we have a meeting sometime this week?

AmrMohamedRashed, 2022-01-05:
@arvuce22  Yes sure I am available Thursday and Friday. What time slot suits you?

AmrMohamedRashed, 2022-01-05:
@arvuce22  here is my email to schedule a meeting

amr.cbnu@gmail.com

LAD021, 2022-03-22:
Hello,
I would like to know whether any updates of docs is available now for mapping ALIGN to new PDK.
Are python and json files of PDK all need to be generated?
I noticed that some .lef with .json files are generated in some intermediate steps.
Can I use my .lef files to continue the flow by translating them to .json format? And is there any guide too?

Thank you in advance.

soneryaldiz, 2022-11-03:
Please refer to [the documentation](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/pdks/PDK_Abstraction_Guide.pptx)

closed 2022-11-03 by soneryaldiz