ALIGN-analoglayout/ALIGN-public #822: Flip Agnostic Sequence Pair Search

soneryaldiz, 2021-10-20:
Given a sequence pair, the flipped versions can either be automatically cached during search or the search algorithm could jump from one another:
```
( pos , neg ) : original sequence
( neg , pos ) : floorplan mirrored along X axis
( gen , sop ) : floorplan mirrored along Y axis
( sop,  gen ) : floorplan mirrored both X and Y axis
```
where sop (gen) is the reversed version of pos (neg). This reduces the search space to one fourth as all of the versions have the same area and hpwl cost.

soneryaldiz, 2021-10-20:
@854768750 , @srini229 fyi.

sapatnekar, 2021-10-21:
The above is not wrong if the objective is area minimization: you have the same bounding box even after flipping. It messes up wire length.

There are some old papers on compaction with wire length minimization. If these are easy to implement, they may help with wire length in general, in addition to providing the same/similar costs for HPWL under mirroring.

There's a simple "plowing" approach in this work:
https://docs.lib.purdue.edu/cgi/viewcontent.cgi?article=1954&context=cstech
Another heuristic is listed in:
https://ieeexplore.ieee.org/abstract/document/1585636

For completeness: as I recall, the precise formulation is a network flow (and I don't know what would happen to that with symmetry constraints). I don't recall the exact reference: I think it is ref [8] of this paper 
https://dl.acm.org/doi/pdf/10.1145/240518.240660
Ref [8] should be available as a hardcopy in our library, but isn't online. I'm guessing we may want to go with something simpler anyway.

soneryaldiz, 2021-11-12:
Closing this issue for now due to higher priority issues.

closed 2021-11-12 by soneryaldiz