ALIGN-analoglayout/ALIGN-public #850: Guardring flow fails

soneryaldiz, 2021-11-04:
Flow fails on telescopic_ota_guard_ring due to a seg fault. To reproduce:

`pytest --runnightly -k telescopic_ota_guard_ring --log-cli-level DEBUG -v -s`

```
align.pnr.toplevel INFO : Running guardring flow
INFO     align.pnr.toplevel:toplevel.py:327 Running guardring flow
Fatal Python error: Segmentation fault
```

fyi, @854768750 , @srini229 .

srini229, 2021-11-04:
@soneryaldiz The failure is because of absence of any pins on the guard ring in the left file generated by the primitive generator. I discussed with @arvuce22 and checked in a fix to the branch fix/guard_ring. It has a DRC now which requires fix in the power net router. The location of the issue is a via pad shown in the figure below:
[image]
r.

arvuce22, 2021-11-12:
@Lastdayends There is a minimum length DRC for M1 in  "telescopic_ota_guard_ring" testcase. Could you please check it in the following branch#862

Lastdayends, 2021-11-13:
@arvuce22 @soneryaldiz 

This should not be a minimum length error. As this M1 is touching its pins in the guard ring. Its pin is a big rectangle. 

[image]

The error reported in the intel checker should be wrong, as the guard ring is missing in the *.json file.

soneryaldiz, 2022-01-18:
Closing this issue as it is not considered critical.

closed 2022-01-18 by soneryaldiz