ALIGN-analoglayout/ALIGN-public #851: Horizontal and Vertical Placement Grid per Block

soneryaldiz, 2021-11-04:
Allow each block layout (e. g. a primitive or an existing circuit) to specify a horizontal and vertical grid for placement. The placement grid is to be specified by an offset and pitch (as in `grid.py::class SingleGrid`). These grids could be specified in JSON format (similar to `generators.py::class Region`) as follows:
```java
{
  'h_grid': {'offset':<value>, 'pitch':<value>}
  'v_grid': {'offset':<value>, 'pitch':<value>}
}
```
where all values are in PDK unit scale.

Example: Single row height standard cell where poly pitch is 50, row height is 300 and PDK scale factor is 10:
```java
{
  'h_grid': {'offset':0, 'pitch':500}
  'v_grid': {'offset':0, 'pitch':3000}
}
```

The algorithm should place a block only on a legal grid position as shown in the examples below. The notch is the lower left of the ori where the notch indicates the lower left corner of the block:

[image]

soneryaldiz, 2021-11-04:
@stevenmburns , is the specification above sufficient?
@854768750 fyi.

stevenmburns, 2021-11-05:
No, I think we need two additions.
There can be more than one pitch offset pair for horizontal (and vertical). (Constraints from two different layers AND-ed together.)
There can also be more than one offset that is legal for a particular pitch (Due to non-uniform grid. These should be OR-ed together.)
My suggestion is something like this:
```
[
  {'type': 'horiz', 'offsets':[0], 'pitch':500},
  {'type': 'horiz', 'offsets':[0,250,350], 'pitch':600},
  {'type': 'vert', 'offsets':[0], 'pitch':3000}
]
```

For the horizontal global offset (oX) we would need this formula to hold:
```
 (oX % 500) in [0] and (oX % 600) in [0,250,350]
```
and
for the vertical global offset (oY) we would need this formula to hold:
```
 (oY % 3000) in [0]
```

We have to also consider when a cell is flip. We might need a separate set of formulas for when a cell is flipped. Let me think about that. We might also need the case when it is illegal to flip a cell, in which case the 'offsets' list could be empty.

stevenmburns, 2021-11-05:
There are certainly going to be cases where we don't want to allow flips so I think this makes sense:
[
  {'type': 'horiz', 'sY': [1,-1], 'offsets':[0], 'pitch':500},
  {'type': 'horiz', 'sY': [1, -1], 'offsets':[0,250,350], 'pitch':600},
  {'type': 'vert', 'sX': [1, -1], 'offsets':[0], 'pitch':3000}
]
'''
This allows both flipping the Y and X coordinates. If you don't want flipping or only want flipping then use the sets [1] and [-1], respectively.

I haven't figured out how to turn these ideas into a formula yet. I think we want the different flippings to be an OR I might need a major overhaul to the spec above. Still thinking.

soneryaldiz, 2021-11-05:
@stevenmburns , I agree on whether to allow horizontal and vertical flipping or not. 

On the grid, yes, there could be multiple layers that determine the grid. My intention was to keep the specification for the downstream engine as simple as possible instead of asking the downstream engine to calculate what the least common multiple is. Let's assume the following two layers: 
`m1, offset=0, pitch of the repeating pattern=100` (could be uniform or non-uniform)
`m3, offset=0, pitch of the repeating pattern=160` (could be uniform or non-uniform)  
If a circuit has wires in both m1 & m3 and mirroring allowed, then the spec for downstream could be simplified to:
`{'h_grid': {'offset': 0, 'pitch':800, 'sX': True, 'sY': True}}`

Can you give an example for having the same pitch with multiple offsets?

stevenmburns, 2021-11-05:
How about this one. You have a repeating height of 11 and you want to allow a cell at position 2 or at position 7 but not anywhere else.
```
------------------------ 11

xxxxxxxxxx
xxxxxxxxxx             7

xxxxxxxxxx
xxxxxxxxxx               2

--------------------------- 0
```

stevenmburns, 2021-11-05:
We do have to figure out who this spec if for. Some of what I'm describing if for the convenience of the spec writer and can probably be expanded out in to a semantically simpler but likely very large description. Let's work first on what we want for the  person writing the spec:
-- I think we should be able to write specs for different layers and have them resolved for us. If the m1 layer provides a constraint and the m3 layer provides a constraint and row sites provide a constraint, then we should be able to write them separately and have the be ANDed together by the tool. (The might create an LCM, etc.) We can provide this expansion routine so that downstream tools don't need to know how to do this
-- We want to be able to express some sort of disjunction: this configuration is legal, and so is this one,  etc. We could enumerate all of the legal configurates with an individual offset, pitch, and legal fliip, but that will be somewhat burdonsome to the spec writer.

So maybe this is reasonable:
```
[
  {'type': 'horiz', [{'sY': [1,-1], 'offsets':[0], 'pitch':500}],
  {'type': 'horiz', [{'sY': [1, -1], 'offsets':[0,250,350], 'pitch':600}],
  {'type': 'vert', [{'sX': [1, -1], 'offsets':[0], 'pitch':3000}]
]
```
This means the same thing as the Cartesian product of the sublists: 
```
[
  {'type': 'horiz', [{'sY': [1], 'offsets':[0], 'pitch':500},{'sY': [-1], 'offsets':[0], 'pitch':500],
  {'type': 'horiz', [{'sY': [1], 'offsets':[0], 'pitch':600}, {'sY': [1], 'offsets':[250], 'pitch':600}, {'sY': [1], 'offsets':[350], 'pitch':600}, {'sY': [-1], 'offsets':[0], 'pitch':600},{'sY': [-1], 'offsets':[250], 'pitch':600}, {'sY': [-1], 'offsets':[350], 'pitch':600}],
  {'type': 'vert', [{"sX': [1], 'offsets':[0], 'pitch':3000}, {"sX': [-1], 'offsets':[0], 'pitch':3000}]
]
```
Now the two horizontal grid constraints (on Y) can also be expanded out, using the LCM, and collapsed into a single disjunction of constraints. So there will be a pitch 3000 and 12 possible terms, many o which disappear of this example. But in general, this is what I think we need. 

We could send the flat version to the downstream tools but we will need multiple choices per pitch.
Let me work on an example that is more interesting.

soneryaldiz, 2021-11-05:
Thank you for the example. I am assuming that this spec can be programmatically generated for a given layout or within a primitive generator. I can see that generating a set of clauses and translating them to a flattened version could make programming easier. Yet is it okay to start with the flattened/simplified version so that placement engine can start being developed?
```java
{
  'h_grid': {'offset':<list of int>, 'pitch':<int>, 'sX': <bool>, 'sY': <bool>}
  'v_grid': {'offset':<list of int>, 'pitch':<int>, 'sX': <bool>, 'sY': <bool>}
}
``` 
We can in parallel or later work out the basic clauses (following your syntax) and the translation logic.

stevenmburns, 2021-11-05:
I don't know why sX and sY are bools. I want them to be lists that can  contain -1, 1, or both. And you only need sY for the h_grid and sX for the v_grid.

soneryaldiz, 2021-11-05:
sX/sY being bool meant to state whether it is okay to flip the cell or not. You are generalizing this with a case where only flipped version is allowed. I guess there is no harm in this generalization.
```java
{
  'h_grid': {'offset':<list of int>, 'pitch':<int>, 'sY': <list of +1,-1>}
  'v_grid': {'offset':<list of int>, 'pitch':<int>, 'sX': <list of +1,-1>}
}
```

stevenmburns, 2021-11-05:
And if you want the flattened version you need a list of dictionarys (the list of alternative positions). So this:
```
{
  'h_grid': [ {'offset':<list of int>, 'pitch':<int>, 'sY': <list with elements -1 and 1>}, ... ], 
  'v_grid':  [ {'offset':<list of int>, 'pitch':<int>, 'sX': <list with elements -1 and 1>}, ... ],
}
```
The offset lists can contain one element as can the sY and sX lists when this has been flattened. And the pitch can be common to all the terms. So it could be this instead
```
{
  'h_grid': {'pitch": <int>, [ {'offset':<list of int>,  'sY': <list with elements -1 and 1>}, ... ]}, 
  'v_grid':  {'pitch': <int>, [ {'offset':<list of int>, 'sX': <list with elements -1 and 1>}, ... ],
}
```
or
```
{
  'h_grid': {'pitch": <int>, [ {'offset':<int>,  'sY': <-1 or 1>}, ... ]}, 
  'v_grid':  {'pitch': <int>, [ {'offset':<int>, 'sX': <-1 or 1>}, ... ],
}
```
but this is further away from the input spec.
The input spec should be now, maybe:
```
[
  {'type': 'h_grid', 'pitch': <int>, [ {'offset':<list of int>,  'sY': <list with elements -1 and 1>}, ... ]},
  ...a second h_grid...
  {'type': 'v_grid', 'pitch': <int>, [ {'offset':<list of int>,  'sX': <list with elements -1 and 1>}, ... ]},
  ...a second v_grid...
]
```

This is my proposal for the flat scheme because it can use the same schema as the input spec:
```
[
  {'type': 'h_grid', 'pitch': <int>, [ {'offset': [<int>],  'sY': [<-1 or 1>]}, ... ]},
  {'type': 'v_grid', 'pitch': <int>, [ {'offset': [<int>],  'sX': [<-1 or 1>]}, ... ]}
]
```

stevenmburns, 2021-11-05:
Here is a more interesting example of multiple pitches and multiple offsets.
If we allow (2 or 4) mod 6 with one constraint, and (2 or 3) mod 5 with another constraint, then that gets expanded to:
(2 or 8 or 22 or 28) mod 30.

soneryaldiz, 2021-11-05:
How about we simplify and JSON'ify the schema as follows?
```java
[
  {'type': 'h_grid', 'pitch': <int>, 'os': [{'offset': <int>, 'scale': <-1 or 1>}, ... ]},
  {'type': 'v_grid', 'pitch': <int>, 'os': [{'offset': <int>, 'scale': <-1 or 1>}, ... ]}
]
```

Example for a single-row cell with 600 row height, 50 poly pitch and flip-able in both directions:
```java
[
  {'type': 'h_grid', 'pitch': 1200, 'os': [{'offset': 0, 'scale': 1}, {'offset': 0, 'scale': -1}],
  {'type': 'v_grid', 'pitch': 50, 'os': [{'offset': 0, 'scale': 1}, {'offset': 0, 'scale': -1}]},
]
```

soneryaldiz, 2021-11-05:
os is basically a list of conditions that can be OR'ed.

stevenmburns, 2021-11-05:
No, because I still think we want the expressiveness of using offset lists and scale lists during specification and we want the schemeas to be the same in both specification and flattened usage. The down stream tool is not going to care too much about unwrapping a couple of lists (and checking that there is exactly one element in it.)

I do like your name changes, though. How about:
```
[
  {'type': 'h_grid', 'pitch': <int>, 'ored_terms': [{'offsets': [<int>], 'scalings': [<-1 or 1>]}, ... ]},
  {'type': 'v_grid', 'pitch': <int>, 'ored_terms': [{'offsets': [<int>], 'scalings': [<-1 or 1>]}, ... ]}
]
```

soneryaldiz, 2021-11-09:
Notes from discussion with Steve:
If the circuit is hierarchical, we need to consider the placement grid for the intermediate level of hierarchy as well. For example, 4-bit flash ADC consists of comparators. Once the comparator subcircuit is placed and routed, the legal placement grid for the comparator instances at ADC level needs to be calculated.

closed 2021-12-22 by soneryaldiz