ALIGN-analoglayout/ALIGN-public #854: Discussion: How to use the new placement injection facility

stevenmburns, 2021-11-05:
Thanks to @854768750 we can now inject a placement into the flow from a placement verilog json file merged into master.

It is working one three test cases, one flat and two that have some hierarchy.
@soneryaldiz @parijatm We should probably try to use it on more test cases.

I'm using the existing --reference_placement_verilog_json command line argument. If you set this,
then it runs the placer in a mode where all it does is take the concrete template names, translations, and scalings from the JSON file and creates a placement datastructure.

Here is an example usage:
```
schematic2layout.py ../../examples/high_speed_comparator/ --reference_placement_verilog_json `pwd`/../../examples/high_speed_comparator/HIGH_SPEED_COMPARATOR_0.scaled_placement_verilog.json --gui
```
(Assuming you are in $ALIGN_HOME/work/<something>.)

I've played around changes the placement of blocks in the JSON file. I can move them around and see the result in the GUI.
I can also create constraint violations doing this. I got an ordering violation and an overlap violation.
I didn't seem to get any symmetric block violation when I made a change that violated that. I guess that has not been hardening yet. 

I haven't tried more complex hierarchical blocks that might contain different concrete_template_names for the same abstract_template_name. I'll see if I can create an example and make sure it works.

If we want to do partial routing, we have some more work to do. I should also make starting a run at the 3_pnr:routing stage work. I'll do that first, but if we want to make progress here we might want to have a dedicated meeting to this early in the week to discuss the next steps.

soneryaldiz, 2021-11-08:
@stevenmburns , 

You are right, symmetric blocks not hardened yet and pending for #807.

Enabling flow start at route would be great. This may allow us to create simple routing test cases for matched/mesh routing enhancements. 

I have an old PR which creates primitives with partial routing. I can resurrect that PR for the next step. Once we have `nvariants` solutions from the placement engine, the next step could be to update the connectivity for each variant if a primitive has partially routed pins. This should require changes only in the leaf definition. 

Can we discuss this during our Tuesday afternoon meeting? or do you prefer to meet later Today?

parijatm, 2022-01-21:
Issue was for information only. Closing

closed 2022-01-21 by parijatm