ALIGN-analoglayout/ALIGN-public #859: Power-Aware Primitive Generation

soneryaldiz, 2021-11-11:
Backside power delivery is a new technology that delivers power to transistors using a special power and eliminates power routing on the front side of the wafer. To leverage this technology, the transistor primitive generator needs to know whether a particular pin is connected to power or ground. This requires compiler to generate unique primitives if a terminal is connected to a power net and propagate this information to the primitive generator.

soneryaldiz, 2021-11-11:
@parijatm , @kkunal1408 , please review. This gets a little tricky when there are multiple ground power pins. Yet  we can start with a single power and ground port.

kkunal1408, 2022-01-17:
Are these separate ports from the D, G, S, B? If yes I can pass these as constraints to the primitives?

soneryaldiz, 2022-01-18:
@kkunal1408 , we are addressing this issue by preprocessing the netlist. Therefore, no changes are required on ALIGN for now.

closed 2022-01-18 by soneryaldiz