ALIGN-analoglayout/ALIGN-public #875: Parameterize multiconnection in primitive generation

parijatm, 2021-11-18:
Currently hardcoded in primitive generator.

parijatm, 2021-11-18:
Need deeper discussion on:
1. Do we want to create so much blockage within the PDK?
2. is it really required to reduce resistance / increase pin area?

meghna09, 2021-12-02:
@soneryaldiz do you prefer primitives with partial routing in all cases instead of primitive routing until M2 or M3?

soneryaldiz, 2021-12-02:
> rimitive routing until M2 or M3?

@meghna09 , this depends on the interconnect stack and the type of the primitive. For example, a single transistor in 2x1 configuration with M1 pins can be easily fully routed by joining adjacent wires. If a PDK has fewer metal layers, then partial routing is preferable so that the layout is route-able.

closed 2022-02-03 by kkunal1408