ALIGN-analoglayout/ALIGN-public #876: Formalize Interface Between ALIGN and Primitive Generators

soneryaldiz, 2021-11-18:
Proposed Tasks:
- [ ] Move primitive generator entirely to a PDK module. One can simply copy the code if necessary for a new PDK
- [ ] Define a scalable interface between ALIGN core and PDK module that generalizes to arrays with 1-n devices and varying device sizes
- [ ] Compiler to query PDK if it can generate arbitrary arrays

As ALIGN is being adopted for multiple PDK's, this would be a good time to standardize as the interface may not be backward compatible.

kkunal1408, 2022-01-18:
Use generator constraint to generate primitive parameters.

soneryaldiz, 2022-01-22:
@parijatm , @kkunal1408, @arvuce22 , is this interface 

```python
from abc import ABC
class PDK(ABC):
    @abstractmethod
    def mos_generator(circuit: subcircuit, max_variants: int) -> (abstract_name: str, variants: dict[str, canvas])
         pass
```

Notes:
- circuit must match a template in basic_templates.sp (non-hierarchical templates) supported by the PDK
- each element in the circuit points is an instance of a Model not subcircuit
- circuit may have constraints that are soft and intended for primitive generator
- each key in variants is a concrete name

kkunal1408, 2022-01-25:
If we want to generate a single nmos, do we still need to pass a circuit definition?

kkunal1408, 2022-01-26:
Thanks for all your time today. I appreciate multiple thoughts on the problem.  We still need more discussion on this and I am trying to summarize our discussion here from my point of view.

Original problem: [generate_primitive_lef](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/align/primitive/main.py) definition looks hacky. We have multiple if-else statements inside it based on different PDK, device names (cap/res/transistor/subcircuit), device parameters (nfin, nf, m, cap, values). Can we formalize this?

# Soner's solution: defined previously (using subcircuit)

# Kunal's question: 
- subcircuit and models definition are different.
- Models are parameterized
- subcircuits have some associated parameters in form of constraints as well as instance-level parameters.

```python
      Library:
	      Subckt1:
		      Instances
                        [
                        xI1 → subckt2 +param1+pins
                        xI2 → subckt2_1 +param2+pins
                        M0 → Model1 + param +pins
                        M1 → Model1 + param +pins
                        ]
            Model1
            Subckt2 (CM_pair):
	            const{CS, parallel=2, aspect_ratio:1:2}
	            M1 → Model + param1 +pins
                    M2 → Model + param1 +pins
            
            Subckt2 (CM_pair_1): CM_pair_1 
	            const{CS, parallel=Pin1:4, aspect_ratio:2:3}
	            M1 → Model_rf + param2 +pins
            M2 → Model_rf + param2 +pins

```
# Parijat's proposed solution:
- Instead of having instance-level parameters, we can convert them to circuit-level parameters.

We discussed this in detail and these are some still unanswered questions:
 - A large number of parameters for subcircuit?
 - Still need a translation from spice definition to generator specific parameter definition.
 - How to associate the instance name to a specific subcircuit layout (aka - abstract name)?

soneryaldiz, 2022-01-26:
@kkunal1408 , thank you for posting the minutes here.

soneryaldiz, 2022-11-10:
This issue has mostly been addressed.

closed 2022-11-10 by soneryaldiz