ALIGN-analoglayout/ALIGN-public #887: Gaps in Power Grid

soneryaldiz, 2021-12-22:
@soneryaldiz 

Actually, there are blockages in the figure.  You might think metal track 1 and 2 should exist, however, the routing track 1 is treated as an blockage in the view of power grid. 
(As routing track 1 is between metal track 1 and 2, from the view of power grid, there is something between metal track 1 and 2, so it is blockage.) 

[image]

_Originally posted by @Lastdayends in https://github.com/ALIGN-analoglayout/ALIGN-public/issues/884#issuecomment-998267572_

soneryaldiz, 2021-12-22:
@Lastdayends 

Are you saying that power grid is not constructed if there exists a signal wire segment in the corresponding power grid cell?

This appears to be quite a conservative strategy. What was the motivation for this?

Does the power tail router have the same blockage concept? Could you please confirm? If yes, this could explain the opens that we observe.

soneryaldiz, 2021-12-27:
Can we try the following on an experimental PR?
1) Draw power/ground grid unless the track on the grid is blocked
2) Export the blockage maps for all terminals even if a terminal is routed successfully to power grid. This could help us to find any issues in labeling blockages.

Lastdayends, 2021-12-27:
@

> @Lastdayends
>
> Are you saying that power grid is not constructed if there exists a signal wire segment in the corresponding power grid cell?
>
> This appears to be quite a conservative strategy. What [...]

Yes, this strategy is for both signal and power routing. Since all our routing is grid-based routing, all internal metal will be mapped on to grid. (Grid are generated based on metal pitches.) Any internal metals will be transferred on to grid as the way below:

[image]

There are two cases. In the figure above, circles is the grid. Red circles mean the grid is inactivated.
1. Internal metal 1 is on grid, then grids inside a metal box will be inactivated. The length of metal box is 2*end_to_end space + length of metal 1, width of the box is the width of metal 1. 
2. Internal metal 2 is off grid, then grids inside a metal box will be inactivated. The length of metal box is 2*end_to_end space + length of metal 1, width of the box is the pitches.

Lastdayends, 2021-12-27:
> Can we try the following on an experimental PR?
> 
> 1. Draw power/ground grid unless the track on the grid is blocked
> 2. Export the blockage maps for all terminals even if a terminal is routed successfully to power grid. This could help us to find any issues in labeling blockages.

I can make a minor update for power grid construction. Let me try it frist.

soneryaldiz, 2021-12-27:
@Lastdayends , let's bring this to the next team meeting and get to the details.

soneryaldiz, 2022-01-14:
This is complete.

closed 2022-01-14 by soneryaldiz