ALIGN-analoglayout/ALIGN-public #898: DRC errors & guard ring

AmrMohamedRashed, 2022-01-02:
I have some notes over the generated layout:
First, the generated layout is not symmetry with respect to a global point/axis.
Second, the guard ring is very small and generates a latch-up DRC error. 
Third, There are some layers are not mapped to the new PDK and can not be deleted.
Lastly, how can I modify "poly" & "Active" overlap parameter and "active" & "Nwell" spacing or enclosure  ? 

Thank you in advance.

AmrMohamedRashed, 2022-01-18:
Hi @kkunal1408 
1. I am using my own example a double tail sense amplifier with Bulk65nm_Mock_PDK
2. I am controlling Guard Ring width by changing activebWidth
{
            "Layer": "Active",
            "GdsLayerNo": 52,
            "GdsDatatype": {
                "Draw": 0
            },
	    "active_enclosure": 42,
            "activebWidth": 168,
            "activebWidth_H": 180,
	    "activePolyExTracks": 2
        },
3. I met with Arvind and he told me how to delete the layers I am not using.
4. Arvind solved the issue of poly and active overlapping.

kkunal1408, 2022-01-18:
@AmrMohamedRashed That's great. If you can add a pull request with your circuit example, I can try to debug the symmetry constraints.

AmrMohamedRashed, 2022-01-18:
@kkunal1408 okay I will do and I will tag you.
thank you.

closed 2022-01-18 by AmrMohamedRashed

arvuce22, 2022-01-18:
@kkunal1408 @AmrMohamedRashed 
I have added the double tail sense amplifier in #917: https://github.com/ALIGN-analoglayout/ALIGN-public/tree/fix/bulkPDK/examples/double_tail_sense_amplifier
@AmrMohamedRashed you can check the example and change it if required

kkunal1408, 2022-01-19:
@AmrMohamedRashed Can you check the layout with the updated master? This branch will also work as I merged master in this branch? You would need one more constraint for the layout to be perfectly symmetric. 
{"constraint": "MergeParallelDevices", "isTrue": false}.
In the default flow, we merge parallel devices for area optimization. In your case, the layout looks better without merging the tail transistor.