ALIGN-analoglayout/ALIGN-public #901: Intermittent failures on 'telescopic_ota_with_bias'

stevenmburns, 2022-01-05:
@srini229 @854768750 @Lastdayends This circuit is sometimes causing failures (every third for me) on the 'merge' level CI. I suspect that it is triggering a memory error.
You can run it standalone as (assuming $ALIGN_WORK_DIR is set to $ALIGN_ROOT/work):
```
mkdir -p work/telescopic_ota_with_bias
schematic2layout.py ../../examples/telescopic_ota_with_bias/ -p ../../pdks/FinFET14nm_Mock_PDK/
```
or through pytest as:
```
CI_LEVEL=merge pytest -vv -k telescopic_ota_with_bias --max-errors 20 -- tests/integration
```
Here is my valgrind run:
```
PYTHONMALLOC=malloc valgrind --tool=memcheck -- schematic2layout.py ../../examples/telescopic_ota_with_bias/ -p ../../pdks/FinFET14nm_Mock_PDK/
```
with (partial---it is still running) output:
```
==23063== Memcheck, a memory error detector
==23063== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==23063== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==23063== Command: /home/smburns/DARPA/ALIGN-public/.venv/bin/schematic2layout.py ../../examples/telescopic_ota_with_bias/ -p ../../pdks/FinFET14nm_Mock_PDK/
==23063== 
==23063== Conditional jump or move depends on uninitialised value(s)
==23063==    at 0x5DE0DB7: __wcsnlen_avx2 (strlen-avx2.S:264)
==23063==    by 0x5D0ECA1: wcsrtombs (wcsrtombs.c:104)
==23063==    by 0x5C94C40: wcstombs (wcstombs.c:34)
==23063==    by 0x5885AF: ??? (in /usr/bin/python3.8)
==23063==    by 0x61AAAE: ??? (in /usr/bin/python3.8)
==23063==    by 0x61AA23: _Py_wreadlink (in /usr/bin/python3.8)
==23063==    by 0x61B3AB: ??? (in /usr/bin/python3.8)
==23063==    by 0x61B22F: _PyPathConfig_Calculate (in /usr/bin/python3.8)
==23063==    by 0x612BC6: _PyConfig_InitPathConfig (in /usr/bin/python3.8)
==23063==    by 0x61144A: PyConfig_Read (in /usr/bin/python3.8)
==23063==    by 0x614135: Py_InitializeFromConfig (in /usr/bin/python3.8)
==23063==    by 0x5E7D50: ??? (in /usr/bin/python3.8)
==23063== 
==23063== Conditional jump or move depends on uninitialised value(s)
==23063==    at 0x5C784E8: internal_utf8_loop (loop.c:298)
==23063==    by 0x5C784E8: __gconv_transform_internal_utf8 (skeleton.c:609)
==23063==    by 0x5D0ECD4: wcsrtombs (wcsrtombs.c:110)
==23063==    by 0x5C94C40: wcstombs (wcstombs.c:34)
==23063==    by 0x5885AF: ??? (in /usr/bin/python3.8)
==23063==    by 0x61AAAE: ??? (in /usr/bin/python3.8)
==23063==    by 0x61AA23: _Py_wreadlink (in /usr/bin/python3.8)
==23063==    by 0x61B3AB: ??? (in /usr/bin/python3.8)
==23063==    by 0x61B22F: _PyPathConfig_Calculate (in /usr/bin/python3.8)
==23063==    by 0x612BC6: _PyConfig_InitPathConfig (in /usr/bin/python3.8)
==23063==    by 0x61144A: PyConfig_Read (in /usr/bin/python3.8)
==23063==    by 0x614135: Py_InitializeFromConfig (in /usr/bin/python3.8)
==23063==    by 0x5E7D50: ??? (in /usr/bin/python3.8)
==23063== 
==23063== Conditional jump or move depends on uninitialised value(s)
==23063==    at 0x5C784F1: internal_utf8_loop (loop.c:303)
==23063==    by 0x5C784F1: __gconv_transform_internal_utf8 (skeleton.c:609)
==23063==    by 0x5D0ECD4: wcsrtombs (wcsrtombs.c:110)
==23063==    by 0x5C94C40: wcstombs (wcstombs.c:34)
==23063==    by 0x5885AF: ??? (in /usr/bin/python3.8)
==23063==    by 0x61AAAE: ??? (in /usr/bin/python3.8)
==23063==    by 0x61AA23: _Py_wreadlink (in /usr/bin/python3.8)
==23063==    by 0x61B3AB: ??? (in /usr/bin/python3.8)
==23063==    by 0x61B22F: _PyPathConfig_Calculate (in /usr/bin/python3.8)
==23063==    by 0x612BC6: _PyConfig_InitPathConfig (in /usr/bin/python3.8)
==23063==    by 0x61144A: PyConfig_Read (in /usr/bin/python3.8)
==23063==    by 0x614135: Py_InitializeFromConfig (in /usr/bin/python3.8)
==23063==    by 0x5E7D50: ??? (in /usr/bin/python3.8)
==23063== 
==23063== Conditional jump or move depends on uninitialised value(s)
==23063==    at 0x5C78538: internal_utf8_loop (loop.c:298)
==23063==    by 0x5C78538: __gconv_transform_internal_utf8 (skeleton.c:609)
==23063==    by 0x5D0ECD4: wcsrtombs (wcsrtombs.c:110)
==23063==    by 0x5C94C40: wcstombs (wcstombs.c:34)
[... 103 lines cut ...]
==23063==    by 0x2613E45D: branch(LP_PROB*, int) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x26117BA8: fathom_branch(LP_PROB*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x261185A8: process_chain(LP_PROB*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x26109BFA: solve(TM_PROB*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x260D35D7: sym_solve(SYM_ENVIRONMENT*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x25F53730: ILP_solver::FrameSolveILPSymphony(design const&, SeqPair const&, PnRDB::Drc_info const&, bool, std::vector<placerDB::point, std::allocator<placerDB::point> > const*) (ILP_solver.cpp:2300)
==23063==    by 0x25F58D17: FrameSolveILP (ILP_solver.h:68)
==23063==    by 0x25F58D17: ILP_solver::GenerateValidSolution(design const&, SeqPair const&, PnRDB::Drc_info const&) (ILP_solver.cpp:2431)
==23063==    by 0x25EFC555: Placer::PlacementCoreAspectRatio_ILP(design&, SeqPair&, ILP_solver&, int, int, int, PnRDB::Drc_info&) (Placer.cpp:735)
==23063==    by 0x25F0212F: Placer::PlacementRegularAspectRatio_ILP(std::vector<PnRDB::hierNode, std::allocator<PnRDB::hierNode> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, PnRDB::Drc_info&) (Placer.cpp:990)
==23063== 
...
```
The first invalid read is probably worth looking at:
```
==23063== Invalid read of size 1
==23063==    at 0x261087D3: generate_children(TM_PROB*, BC_NODE*, BRANCH_OBJ*, double*, int*, char*, int, int*, int) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x2611D937: send_branching_info(LP_PROB*, BRANCH_OBJ*, char*, int*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x2613E45D: branch(LP_PROB*, int) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x26117BA8: fathom_branch(LP_PROB*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x261185A8: process_chain(LP_PROB*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x26109BFA: solve(TM_PROB*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x260D35D7: sym_solve(SYM_ENVIRONMENT*) (in /home/smburns/DARPA/ALIGN-public/align/PnR.cpython-38-x86_64-linux-gnu.so)
==23063==    by 0x25F53730: ILP_solver::FrameSolveILPSymphony(design const&, SeqPair const&, PnRDB::Drc_info const&, bool, std::vector<placerDB::point, std::allocator<placerDB::point> > const*) (ILP_solver.cpp:2300)
```

stevenmburns, 2022-01-05:
Here is the rest of the valgrind run including a crash:
```
PnR.router.Router.RouteWork INFO : GcellGlobalRouter: CMB_NMOS_2_PG0
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Format Issue 
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Please check the net DB in module CMB_NMOS_2_PG0
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Especial the pin DB in subblock X_SCM_NMOS_M0_M1
==23063== Invalid read of size 4
==23063==    at 0x25EAE761: construct<int, int const&> (new_allocator.h:136)
==23063==    by 0x25EAE761: construct<int, int const&> (alloc_traits.h:475)
==23063==    by 0x25EAE761: void std::vector<int, std::allocator<int> >::_M_realloc_insert<int const&>(__gnu_cxx::__normal_iterator<int*, std::vector<int, std::allocator<int> > >, int const&) (vector.tcc:415)
==23063==    by 0x2604909F: push_back (stl_vector.h:948)
==23063==    by 0x2604909F: GlobalGraph::MST(int&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >&, GlobalGrid&) (GlobalGraph.cpp:247)
==23063==    by 0x26049C75: GlobalGraph::FindSTs(GlobalGrid&, int, std::vector<int, std::allocator<int> >&) (GlobalGraph.cpp:31)
==23063==    by 0x2602AF62: GcellGlobalRouter::GcellGlobalRouter(PnRDB::hierNode&, PnRDB::Drc_info&, int, int) (GcellGlobalRouter.cpp:552)
==23063==    by 0x25FDBC53: Router::RouteWork(int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (Router.cpp:50)
==23063==    by 0x25E55808: void pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, Router, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pybind11::name, pybind11::is_method, pybind11::sibling>(void (Router::*)(int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::{lambda(Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)#1}, void, Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pybind11::name, pybind11::is_method, pybind11::sibling>(pybind11::cpp_function::initialize<void, Router, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pybind11::name, pybind11::is_method, pybind11::sibling>(void (Router::*)(int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::{lambda(Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)#1}&&, void (*)(Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::{lambda(pybind11::detail::function_call&)#3}::operator()(pybind11::detail::function_call) const (pybind11.h:84)
==23063==    by 0x25E5B9BF: pybind11::cpp_function::dispatcher(_object*, _object*, _object*) (pybind11.h:767)
==23063==    by 0x4FB287: PyCFunction_Call (in /usr/bin/python3.8)
==23063==    by 0x4F9C28: _PyObject_MakeTpCall (in /usr/bin/python3.8)
==23063==    by 0x5CE735: ??? (in /usr/bin/python3.8)
==23063==    by 0x55A439: _PyEval_EvalFrameDefault (in /usr/bin/python3.8)
==23063==    by 0x55591E: _PyEval_EvalCodeWithName (in /usr/bin/python3.8)
==23063==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==23063== 
==23063== 
==23063== Process terminating with default action of signal 11 (SIGSEGV)
==23063==  Access not within mapped region at address 0x0
==23063==    at 0x25EAE761: construct<int, int const&> (new_allocator.h:136)
==23063==    by 0x25EAE761: construct<int, int const&> (alloc_traits.h:475)
==23063==    by 0x25EAE761: void std::vector<int, std::allocator<int> >::_M_realloc_insert<int const&>(__gnu_cxx::__normal_iterator<int*, std::vector<int, std::allocator<int> > >, int const&) (vector.tcc:415)
==23063==    by 0x2604909F: push_back (stl_vector.h:948)
==23063==    by 0x2604909F: GlobalGraph::MST(int&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >&, GlobalGrid&) (GlobalGraph.cpp:247)
==23063==    by 0x26049C75: GlobalGraph::FindSTs(GlobalGrid&, int, std::vector<int, std::allocator<int> >&) (GlobalGraph.cpp:31)
==23063==    by 0x2602AF62: GcellGlobalRouter::GcellGlobalRouter(PnRDB::hierNode&, PnRDB::Drc_info&, int, int) (GcellGlobalRouter.cpp:552)
==23063==    by 0x25FDBC53: Router::RouteWork(int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (Router.cpp:50)
==23063==    by 0x25E55808: void pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, Router, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pybind11::name, pybind11::is_method, pybind11::sibling>(void (Router::*)(int, PnRDB::hierNode&,  [... cut ...]
==23063==                         stdstring          : 9,800 bytes in 259 blocks
==23063==                         newarray           : 592 bytes in 37 blocks
==23063==         suppressed: 0 bytes in 0 blocks
==23063== Rerun with --leak-check=full to see details of leaked memory
==23063== 
==23063== For counts of detected and suppressed errors, rerun with: -v
==23063== Use --track-origins=yes to see where uninitialised values come from
==23063== ERROR SUMMARY: 5725 errors from 7 contexts (suppressed: 0 from 0)
Segmentation fault
```

srini229, 2022-01-05:
@stevenmburns The invalid reads are from the symphony solver. I saved the ILP problem solved during placement into a LP file and solved the problem using the standalone version of "SYMPHONY" solver attached to valgrind. I still get the ``Invalid read of size 1``. This is not causing any memory leaks though. We can maybe pass this as a test to the SYMPHONY developers.

stevenmburns, 2022-01-05:
It is probably not the issue then. There is a bigger problem with the second error in the global router. Perhaps @854768750 or @Lastdayends can take a look. From above:
```
PnR.router.Router.RouteWork INFO : GcellGlobalRouter: CMB_NMOS_2_PG0
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Format Issue 
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Please check the net DB in module CMB_NMOS_2_PG0
PnR.router.GcellGlobalRouter.GcellGlobalRouter ERROR : Especial the pin DB in subblock X_SCM_NMOS_M0_M1
==23063== Invalid read of size 4
==23063==    at 0x25EAE761: construct<int, int const&> (new_allocator.h:136)
==23063==    by 0x25EAE761: construct<int, int const&> (alloc_traits.h:475)
==23063==    by 0x25EAE761: void std::vector<int, std::allocator<int> >::_M_realloc_insert<int const&>(__gnu_cxx::__normal_iterator<int*, std::vector<int, std::allocator<int> > >, int const&) (vector.tcc:415)
==23063==    by 0x2604909F: push_back (stl_vector.h:948)
==23063==    by 0x2604909F: GlobalGraph::MST(int&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >&, GlobalGrid&) (GlobalGraph.cpp:247)
==23063==    by 0x26049C75: GlobalGraph::FindSTs(GlobalGrid&, int, std::vector<int, std::allocator<int> >&) (GlobalGraph.cpp:31)
==23063==    by 0x2602AF62: GcellGlobalRouter::GcellGlobalRouter(PnRDB::hierNode&, PnRDB::Drc_info&, int, int) (GcellGlobalRouter.cpp:552)
==23063==    by 0x25FDBC53: Router::RouteWork(int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (Router.cpp:50)
==23063==    by 0x25E55808: void pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, Router, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pybind11::name, pybind11::is_method, pybind11::sibling>(void (Router::*)(int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::{lambda(Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)#1}, void, Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pybind11::name, pybind11::is_method, pybind11::sibling>(pybind11::cpp_function::initialize<void, Router, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pybind11::name, pybind11::is_method, pybind11::sibling>(void (Router::*)(int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::{lambda(Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)#1}&&, void (*)(Router*, int, PnRDB::hierNode&, PnRDB::Drc_info&, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&)::{lambda(pybind11::detail::function_call&)#3}::operator()(pybind11::detail::function_call) const (pybind11.h:84)
==23063==    by 0x25E5B9BF: pybind11::cpp_function::dispatcher(_object*, _object*, _object*) (pybind11.h:767)
==23063==    by 0x4FB287: PyCFunction_Call (in /usr/bin/python3.8)
==23063==    by 0x4F9C28: _PyObject_MakeTpCall (in /usr/bin/python3.8)
==23063==    by 0x5CE735: ??? (in /usr/bin/python3.8)
==23063==    by 0x55A439: _PyEval_EvalFrameDefault (in /usr/bin/python3.8)
==23063==    by 0x55591E: _PyEval_EvalCodeWithName (in /usr/bin/python3.8)
==23063==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
```

arvuce22, 2022-01-06:
@stevenmburns @Lastdayends There were some issues in the netlists. We have fixed all the netlists (i.e., examples) in #895. Now all these issues should get resolved.

arvuce22, 2022-01-06:
@stevenmburns @Lastdayends Now I am not facing any issues in "telescopic_ota_with_bias". Please let me know if it does not work at your end.

stevenmburns, 2022-01-06:
@parijatm  @soneryaldiz @kkunal1408 @arvuce22 This problem has disappeared but a new one has emerged on the same block. It is still failing the master merge check.
When run in standalone mode, it sometimes passes and sometimes produces this error:
```
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/work/telescopic_ota_with_bias$ schematic2layout.py ../../examples/telescopic_ota_with_bias/ -p ../../pdks/FinFET14nm_Mock_PDK/
align.main INFO : Running flow steps ['1_topology', '2_primitives', '3_pnr:prep', '3_pnr:place', '3_pnr:route', '3_pnr:check']
align.main INFO : READ file: /home/smburns/DARPA/ALIGN-public/examples/telescopic_ota_with_bias/telescopic_ota_with_bias.sp subckt=TELESCOPIC_OTA_WITH_BIAS, flat=0
align.compiler.compiler INFO : Starting topology identification...
align.compiler.user_const INFO : Reading constraints for telescopic_ota_with_bias
align.compiler.preprocess INFO : stacking ['M8', 'M8S'] 2
align.compiler.preprocess INFO : stacking ['M4', 'M4S'] 2
align.compiler.preprocess INFO : stacking ['M2', 'M2S'] 2
align.compiler.preprocess INFO : stacking ['M10', 'M10S'] 2
align.compiler.preprocess INFO : stacking ['M9', 'M9S'] 2
align.compiler.preprocess INFO : stacking ['M0', 'M0S'] 2
align.compiler.preprocess INFO : stacking ['M7', 'M7S'] 2
align.compiler.preprocess INFO : stacking ['M17', 'M17S'] 2
align.compiler.preprocess INFO : stacking ['M14', 'M14S'] 2
align.compiler.preprocess INFO : stacking ['M12', 'M12S'] 2
align.compiler.preprocess INFO : stacking ['M11', 'M11S'] 2
align.compiler.preprocess INFO : stacking ['M13', 'M13S'] 2
align.compiler.preprocess INFO : stacking ['M15', 'M15S'] 2
align.compiler.preprocess INFO : stacking ['M5', 'M5S'] 2
align.compiler.preprocess INFO : stacking ['M16', 'M16S'] 2
align.compiler.preprocess INFO : stacking ['M6', 'M6S'] 2
align.compiler.preprocess INFO : stacking ['M3', 'M3S'] 2
align.compiler.preprocess INFO : stacking ['M1', 'M1S'] 2
align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/smburns/DARPA/ALIGN-public/align/cmdline.py", line 161, in parse_args
    return schematic2layout(**vars(arguments))
  File "/home/smburns/DARPA/ALIGN-public/align/main.py", line 300, in schematic2layout
    primitives, ckt_data = generate_hierarchy(netlist, subckt, topology_dir, flatten, pdk_dir, uniform_height)
  File "/home/smburns/DARPA/ALIGN-public/align/compiler/compiler.py", line 41, in generate_hierarchy
    primitives, generators = call_primitive_generator(
  File "/home/smburns/DARPA/ALIGN-public/align/compiler/compiler.py", line 175, in call_primitive_generator
    assert generate_primitive_lef(
  File "/home/smburns/DARPA/ALIGN-public/align/primitive/main.py", line 354, in generate_primitive_lef
    assert int(values[device_name]["NFIN"]) == int(values[key]["NFIN"]), f"NFIN should be same for all devices in {name} {device_name}={values[device_name]['NFIN']} {key}={values[key]['NFIN']}"
AssertionError: NFIN should be same for all devices in SCM_NMOS M1=5 M2=10
```
(I added the devices naming reporting.)
Is there anything that could be done non-deterministically in the flow? Shouldn't it always fail or always succeed?

stevenmburns, 2022-01-06:
@kkunal1408 @arvuce22 In looking at this more carefully, it seems that we are getting different parameter values of  the devices in different runs of `generate_primitive_lef`:
This is a failure:
```
align.primitive.main INFO : SMB: NMOS {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': 2}
align.primitive.main INFO : SMB2: {'M11': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': 2}} ['RVT']
align.primitive.main INFO : SMB: SCM_PMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: CMC_S_NMOS_B None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '6', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '6', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: DP_NMOS_B None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '10', 'NF': '14', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '10', 'NF': '14', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: PMOS_S None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: DCL_NMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '3', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: DCL_PMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '3', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: NMOS_S None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: CMC_PMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '4', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '4', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: CMC_S_PMOS_B None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '8', 'NF': '12', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '8', 'NF': '12', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: SCM_NMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '4', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '10', 'NF': '10', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
```
and this one is okay:
```
align.primitive.main INFO : SMB: NMOS {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': 2}
align.primitive.main INFO : SMB2: {'M11': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': 2}} ['RVT']
align.primitive.main INFO : SMB: SCM_PMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: CMC_S_NMOS_B None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '6', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '6', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: DP_NMOS_B None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '10', 'NF': '14', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '10', 'NF': '14', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: PMOS_S None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: DCL_NMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '3', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: DCL_PMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '3', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: NMOS_S None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '10', 'NF': '10', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: CMC_PMOS None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '4', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}, 'M2': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '12', 'NF': '4', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: CMC_S_PMOS_B None
[... cut ...] ', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: NMOS_S None
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
align.primitive.main INFO : SMB: NMOS {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}
align.primitive.main INFO : SMB2: {'M1': {'W': '2.7E-07', 'L': '2E-08', 'NFIN': '5', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '2'}} ['RVT']
```

Lastdayends, 2022-01-07:
> @stevenmburns @Lastdayends Now I am not facing any issues in "telescopic_ota_with_bias". Please let me know if it does not work at your end.

Good. Thanks for the update!

stevenmburns, 2022-01-17:
Fixed by #910

closed 2022-01-17 by stevenmburns