ALIGN-analoglayout/ALIGN-public #902: Several tests in tests/pdks failing during CI on merge to master

stevenmburns, 2022-01-05:
@kkunal1408 @arvuce22 @parijatm When I run the merge to master CI level manually, I get several errors from `tests/pdks`.
Here is a list of the failures and how to duplicate the errors:
```
FAILED tests/pdks/test_cap.py::test_cap_full[FinFET14nm_Mock_PDK-n4] - json.decoder.JSONDecodeError: Extra data: line 1685 column 9 (ch...
FAILED tests/pdks/test_cap.py::test_cap_full[FinFET14nm_Mock_PDK-n20] - json.decoder.JSONDecodeError: Extra data: line 3840 column 3 (c...
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X1-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X3-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X2-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X3-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X2-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X4-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X3-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X1-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X2-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X1-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X4-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X2-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X1-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X1-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X2-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X4-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X4-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X4-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X4-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X3-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-NMOS-n12-X2-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X3-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X1-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[FinFET14nm_Mock_PDK-CM-PMOS-n12-X3-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X1-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X1-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X1-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X2-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X1-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X2-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X2-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X4-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X2-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X4-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X2-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X3-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X2-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X3-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X3-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X3-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X3-Y2] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X4-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X3-Y3] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X1-Y1] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X4-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X1-Y2] - NotImplementedError: Unrecognized primitive CM_NMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-PMOS-n12-X4-Y1] - NotImplementedError: Unrecognized primitive CM_PMOS
FAILED tests/pdks/test_mos.py::test_mos_full[Bulk65nm_Mock_PDK-CM-NMOS-n12-X4-Y3] - NotImplementedError: Unrecognized primitive CM_NMOS
```
To duplicate this:
```
CI_LEVEL='merge' pytest -n 8 -vv --runnightly --maxerrors=20 -- tests/pdks
```

arvuce22, 2022-01-05:
@stevenmburns  These errors are due to modified primitive names. Addressed in #904

stevenmburns, 2022-01-06:
Fixed in #904

closed 2022-01-06 by stevenmburns