ALIGN-analoglayout/ALIGN-public #928: Using ALIGN for VCOs/PLL subcircuits

powergainer, 2022-01-18:
Hello,
Please excuse if this is not the correct place to ask this but I wasn't sure how to contact the repo owners. I am working on some VCO / PLL design in Sky130 and I would like to know 1) whether I can use this ALIGN software to generate a layout for a VCO circuit and other PLL subblocks like the CP etc and 2) maybe there is something I could contribute to this repo while trying to do so?
I have seen some VCO circuits in the circuits database but I may be designing a different VCO topology. So, the summary is, I'm trying to understand if I can make use of this ALIGN software to generate layouts in a quick and efficient way for my real application, whether this is practical in terms of learning curve and actual timescales for trying to do it, and whether I can help or contribute anything to this work in the process.
As mentioned before, if this is not the right place for this conversation I would appreciate if you could indicate to me how or where to talk about this topics.
Thanks and all the best with your project.

arvuce22, 2022-01-27:
@powergainer We are working on Sky130 integration in the ALIGN flow (https://github.com/ALIGN-analoglayout/ALIGN-pdk-sky130). It would be helpful if you can add your netlists to the "examples" directory of the following repo:  https://github.com/ALIGN-analoglayout/ALIGN-pdk-sky130

arvuce22, 2022-04-07:
@powergainer We have integrated Sky130 in the ALIGN in the following repo: https://github.com/ALIGN-analoglayout/ALIGN-pdk-sky130
Closing this issue for now. Please let us know if you face any issues.

powergainer, 2022-04-10:
Hi @arvuce22 I am getting some errors while installing ALIGN. I can attach the full Log if required, but the last lines of the error are like this:

`  × Building wheel for align (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /home/darunix/GitSandboxes/ALIGN/ALIGN-public/general/bin/python /home/darunix/GitSandboxes/ALIGN/ALIGN-public/general/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmpi63a2gql
  cwd: /home/darunix/GitSandboxes/ALIGN/ALIGN-public
  Building wheel for align (pyproject.toml) ... error
  ERROR: Failed building wheel for align
Failed to build align
ERROR: Could not build wheels for align, which is required to install pyproject.toml-based projects
`
Do you know how to fix this? Thanks

arvuce22, 2022-04-11:
@kkunal1408

arvuce22, 2022-04-13:
@kkunal1408 @powergainer Kunal, Can you help in resolving ALIGN installation issue?

kkunal1408, 2022-04-13:
Hi @powergainer 
The failures can be due to any of the prerequisites not being installed properly. We would need more system info such as machine, Cmake and C++ versions. If you can share the log it will help us in debugging it. 

Another installation method is to use already generated wheels. You can skip the wheel generation step by using this latest wheel for Python 3.7/3.8. 
https://output.circle-artifacts.com/output/job/f6b214e7-9e2a-419a-aabf-26a2fda3d781/artifacts/0/export/align-wheelhouse-506498f.tar.gz

You can check some more discussions on #817

arvuce22, 2022-05-12:
@powergainer closing the issue. Please let us know if you are still facing the issue.

closed 2022-05-12 by arvuce22