ALIGN-analoglayout/ALIGN-public #951: Placement Runtime with PlaceOnGrid Constraints

soneryaldiz, 2022-01-26:
PlaceOnGrid constraints significantly slow down placement.

```csh
cd tests/pdk/finfet_pdk
# It takes ~ 40 secs to generate layout w/o PlaceOnGrid
time pytest test_circuits.py::test_cmp_vanilla -vv -s --log-cli-level INFO
# 5 minutes yet no progress on placement
setenv PLACE_ON_GRID t && time pytest test_circuits.py::test_cmp_vanilla -vv -s --log-cli-level INFO
```

soneryaldiz, 2022-01-26:
@854768750 , could you please look into the root cause?

854768750, 2022-01-26:
> @854768750 , could you please look into this?

I will look into this.

soneryaldiz, 2022-01-27:
@854768750 , please try on #954 branch as the constraints are only generated for PMOS on master branch.

854768750, 2022-02-03:
The problem is here that the ilp solver cannot generate a solution as the geometric constraints conflict.
Block 1 has height 2520, block 2 has height 5040, and they are sympair (their centers have the same y value).
Place on grid wants them to be on pitch=2520. And this is impossible. 
The ILP solver should exit with infeasible status but did not.

soneryaldiz, 2022-02-03:
@854768750 , great finding. Which two block are these?

854768750, 2022-02-03:
X_MN11_MP13 and X_MN12_MP14, while different instance indices

I would like to validate the sequence pair before running ILP.

854768750, 2022-02-03:
@srini229 Just let you know we were talking about this issue. The ILP solver does not stop when there is no feasible solution.

soneryaldiz, 2022-02-04:
Added a new test in #965 with properly setup constraints. It takes about:
~ 25 secs w/o PlaceOnGrid
~ 35 secs w/. PlaceOnGrid

closed 2022-02-04 by soneryaldiz