ALIGN-analoglayout/ALIGN-public #966: PlaceOnGrid OR'ing of offset constraints

srini229, 2022-02-04:
@854768750 As per my understanding of the PlaceOnGrid constraint only one of the ORed offset constraints need to satisfied.
The code below is adding a constraint for each of the offset values; meaning all of them have to be simultaneously satisfied. https://github.com/ALIGN-analoglayout/ALIGN-public/blob/bc7cc48a853bba923af02d1bd4676c98aa266331/PlaceRouteHierFlow/placer/ILP_solver.cpp#L1821-L1832

The fix would be to generate an intermediate xoffset<sub>combined</sub> variable as:
[image]

Use the xoffset<sub>combined</sub> in the

854768750, 2022-02-04:
please see the next 7 lines.

 

 
The fix would be to generate an intermediate xoffsetcombined variable as:
 
 
Use the xoffsetcombined in the

854768750, 2022-02-04:
please also see
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/bc7cc48a853bba923af02d1bd4676c98aa266331/PlaceRouteHierFlow/placer/ILP_solver.cpp#L1688-1707

srini229, 2022-02-04:
@854768750 the next seven lines are just implementing 
&Sigma; select<sub>i</sub> = 1
select<sub>i</sub> = {0,1}

The equation:
xoffset<sub>combined</sub> = Σ select<sub>i</sub> * xoffset<sub>i</sub>
is not implemented.

854768750, 2022-02-04:
@srini229 you are right. Thanks. Please see https://github.com/ALIGN-analoglayout/ALIGN-public/pull/965/commits/eec4ed3cdaea5060c7bd4857ddc49a5144a8497d

soneryaldiz, 2022-02-04:
@srini229 , I pushed a commit to #965 which creates multiple ored_terms. Try: `pytest test_placement_grid.py::test_tia_on_grid`

Yet the test passes and does not demonstrate the issue.

854768750, 2022-02-04:
> @srini229 , I pushed a commit to #965 which creates multiple ored_terms. Try: `pytest test_placement_grid.py::test_tia_on_grid`
> 
> Yet the test passes and does not demonstrate the issue.

Yes I just fixed the error that @srini229 mentioned.

soneryaldiz, 2022-02-04:
It looks like @854768750 fixed it before I get to it. 
Regardless, relying on environment variables to turn on/off experimental features turns out to be dangerous as the variable is preserved across tests. Suggestions are welcome.

soneryaldiz, 2022-02-04:
monkeypatch does the job! @srini229 , shall we close this issue now?

closed 2022-02-04 by soneryaldiz