ALIGN-analoglayout/ALIGN-public #982: Exclude net from Placement optimization

854768750, 2022-02-17:
A new constraint: zero-weighted net constraint/weighted net constraint
@soneryaldiz Do you want to add this new feature in placer?

soneryaldiz, 2022-02-18:
@854768750 , we may not need a new feature based on the recent discussions.
- What happens when `criticality` in `NetConst` is set to zero?
- What is the default value of `criticality` (when not specified)?

854768750, 2022-02-18:
`NetConst` is not implemented. It defines order of net routing.

soneryaldiz, 2022-02-18:
Okay, the answer is in `write_constraint.py`. 
@854768750 , is `CritNet` implemented? what is the default `priority` of a net?

854768750, 2022-02-18:
There is no `CritNet`.

kkunal1408, 2022-02-18:
Its defined in PnRDB/ReadConstraint.cpp
"""
else if (constraint["const_name"] == "CritNet") {
      for (int i = 0; i < (int)node.Nets.size(); i++) {
        if (node.Nets.at(i).name == constraint["net_name"]) {
          node.Nets.at(i).priority = constraint["priority"];
          break;
        }
      }
"""

854768750, 2022-02-21:
It is not supported in the later flow. If it is required, please let me know.

854768750, 2022-02-22:
> @854768750 , thank you for the clarification. I think we should continue to retire any dead code that is not completed.
>
> Reducing/zeroing the priority of select nets is required. How about the following?
>  {     "constraint": "SetPriority",     "nets": list(str),     "priority": int }
>
> Example:
>  {     "constraint": "SetPriority",     "nets": "clk_enable",     "priority": 0 }

It is fine.

854768750, 2022-03-24:
Solved in #1010

closed 2022-03-24 by 854768750