ALIGN-analoglayout/ALIGN-public #998: Fix errors in switched capacitor filter

stevenmburns, 2022-03-01:
We have two kinds of error showing up in switched_capacitor_filiter.
- [ ] A routing error (open)
- [ ] A missing pin in a primitive

```
...
PnR.PnRDB.PnRdatabase.MergeLEFMapData ERROR : Block X_M4_M5 pin DB not found in lef file
...
align.cell_fabric.remove_duplicates ERROR : Found errors: SHORT: 0 OPEN: 1 DIFFERENT WIDTH: 14
...
```

To reproduce:
```
mkdir -p work/switched_capacitor_filter
cd work/switched_capacitor_filter
schematic2layout.py ../examples/switched_capacitor_filter
```
We get this log:
```
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/work/switched_capacitor_filter$ schematic2layout.py ../../examples/switched_capacitor_filter/
align.main INFO : Running flow steps ['1_topology', '2_primitives', '3_pnr:prep', '3_pnr:place', '3_pnr:route', '3_pnr:check']
align.main INFO : READ file: /home/smburns/DARPA/ALIGN-public/examples/switched_capacitor_filter/switched_capacitor_filter.sp subckt=SWITCHED_CAPACITOR_FILTER, flat=0
align.compiler.compiler INFO : Starting topology identification...
align.compiler.read_library INFO : Using model file from /home/smburns/DARPA/ALIGN-public/pdks/FinFET14nm_Mock_PDK/models.sp
align.compiler.read_library INFO : Using model file from /home/smburns/DARPA/ALIGN-public/pdks/FinFET14nm_Mock_PDK/models.sp
align.compiler.user_const INFO : Reading constraints for switched_capacitor_filter
align.compiler.match_graph INFO : Existing GroupCaps constraint [GroupCaps(constraint='group_caps', name='c1_c3', instances=List[str](__root__=['C1', 'C3']), unit_cap='Cap_12f', num_units=List(__root__=[3, 5]), dummy=False), GroupCaps(constraint='group_caps', name='c7_c6', instances=List[str](__root__=['C7', 'C6']), unit_cap='Cap_12f', num_units=List(__root__=[3, 5]), dummy=False)] for subckt SWITCHED_CAPACITOR_FILTER
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C1_C3', 'C1', 'C3'] with C1_C3
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C7_C6', 'C7', 'C6'] with C7_C6
align.schema.translator INFO : update constraints of TELESCOPIC_OTA :['M5', 'M4'] with X_M4_M5
align.schema.translator INFO : update constraints of TELESCOPIC_OTA :['M1', 'M2'] with X_M1_M2
align.schema.translator INFO : update constraints of TELESCOPIC_OTA :['M6', 'M7'] with X_M6_M7
align.schema.translator INFO : update constraints of TELESCOPIC_OTA :['M8', 'M9'] with X_M8_M9
align.schema.translator INFO : update constraints of TELESCOPIC_OTA :['M0', 'M3'] with X_M0_M3
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M4', 'M8'] with X_M4_M8
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M0', 'M14'] with X_M0_M14
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M10', 'M7'] with X_M10_M7
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M12', 'M6'] with X_M12_M6
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C0'] with X_C0
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C2'] with X_C2
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C4'] with X_C4
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C5'] with X_C5
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C8'] with X_C8
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['C9'] with X_C9
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M11'] with X_M11
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M3'] with X_M3
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M5'] with X_M5
align.schema.translator INFO : update constraints of SWITCHED_CAPACITOR_FILTER :['M9'] with X_M9
align.schema.translator INFO : update constraints of CMC_S_PMOS :['M1'] with X_M1
align.schema.translator INFO : update constraints of CMC_S_PMOS :['M2'] with X_M2
align.compiler.gen_abstract_name INFO : found group cap instances []
align.compiler.gen_abstract_name INFO : found group cap instances ['C1_C3', 'C7_C6']
align.compiler.gen_abstract_name INFO : group cap instance model='CAP_CC_3_5' name='C1_C3' pins={'PLUS0': 'NET63', 'MINUS0': 'NET67', 'PLUS1': 'NET66', 'MINUS1': 'VOUTN'} parameters=None abstract_name='CAP_CC_3_5'
align.compiler.gen_abstract_name INFO : group cap instance model='CAP_CC_3_5' name='C7_C6' pins={'PLUS0': 'NET62', 'MINUS0': 'NET68', 'PLUS1': 'NET64', 'MINUS1': 'VOUTP'} parameters=None abstract_name='CAP_CC_3_5'
align.compiler.gen_abstract_name INFO : found group cap instances []
align.compiler.compiler INFO : Completed topology identification.
align.pnr.main INFO : Running Place & Route for SWITCHED_CAPACITOR_FILTER top_down ['3_pnr:prep', '3_pnr:place', '3_pnr:route', '3_pnr:check']
align.pnr.cap_placer INFO : Running cap_placer_driver...
PnR.PnRDB.PnRdatabase.MergeLEFMapData ERROR : Block X_M4_M5 pin DB not found in lef file
align.pnr.cap_placer INFO : Starting bottom-up cap placement on CMC_S_PMOS 1
align.pnr.cap_placer INFO : Starting bottom-up cap placement on TELESCOPIC_OTA_PG0 2
align.pnr.cap_placer INFO : Starting bottom-up cap placement on SWITCHED_CAPACITOR_FILTER 0
PnR.cap_placer.Placer_Router_Cap.Common_centroid_capacitor_aspect_ratio INFO : CC Capacitor Placement and Router : SWITCHED_CAPACITOR_FILTER C1_C3
PnR.cap_placer.Placer_Router_Cap.Common_centroid_capacitor_aspect_ratio INFO : CC Capacitor Placement and Router : SWITCHED_CAPACITOR_FILTER C7_C6
align.pnr.cap_placer INFO : Required capacitors: {'CAP_CC_3_5'}
align.pnr.cap_placer INFO : Found lef file for CC capacitor CAP_CC_3_5 CAP_CC_3_5_AspectRatio_3x3
align.pnr.build_pnr_model INFO : Reading LEF from string...
align.pnr.build_pnr_model WARNING : No primitive json file for primitive CAP_CC_3_5_AspectRatio_3x3. Okay if a CC capacitor.
PnR.PnRDB.PnRdatabase.MergeLEFMapData ERROR : Block X_M4_M5 pin DB not found in lef file
[... 71 lines cut ...]
PnR.router.Router.RouteWork INFO : Create power grid: SWITCHED_CAPACITOR_FILTER
PnR.router.Router.RouteWork INFO : Power routing SWITCHED_CAPACITOR_FILTER
align.pnr.checkers INFO : Checking: CMC_S_PMOS
align.pnr.main INFO : OUTPUT json at /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/3_pnr/CMC_S_PMOS_0_0.json
align.pnr.main INFO : OUTPUT gds.json /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/3_pnr/CMC_S_PMOS_0_0.python.gds.json
align.pnr.checkers INFO : Checking: TELESCOPIC_OTA_PG0
align.pnr.main INFO : OUTPUT json at /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/3_pnr/TELESCOPIC_OTA_PG0_0_0.json
align.pnr.main INFO : OUTPUT gds.json /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/3_pnr/TELESCOPIC_OTA_PG0_0_0.python.gds.json
align.pnr.checkers INFO : Checking: SWITCHED_CAPACITOR_FILTER
align.cell_fabric.remove_duplicates WARNING : Layer Cboundary not in OrderedDict([('M1', 'v'), ('M2', 'h'), ('V1', '*'), ('M3', 'v'), ('V2', '*'), ('M4', 'h'), ('V3', '*'), ('M5', 'v'), ('V4', '*'), ('M6', 'h'), ('V5', '*'), ('M7', 'v'), ('V6', '*'), ('M8', 'h'), ('V7', '*'), ('M9', 'v'), ('V8', '*'), ('M10', 'h'), ('V9', '*'), ('M11', 'v'), ('V10', '*'), ('M12', 'h'), ('V11', '*'), ('M13', 'v'), ('V12', '*'), ('M14', 'h'), ('V13', '*'), ('M15', 'v'), ('V14', '*'), ('Poly', 'v'), ('Fin', 'h'), ('Active', 'h'), ('Rvt', 'h'), ('Lvt', 'h'), ('Hvt', 'h'), ('Slvt', 'h'), ('Lisd', 'v'), ('Pc', 'h'), ('Pb', 'h'), ('V0', '*')])
align.cell_fabric.remove_duplicates WARNING : Layer Cboundary not in OrderedDict([('M1', 'v'), ('M2', 'h'), ('V1', '*'), ('M3', 'v'), ('V2', '*'), ('M4', 'h'), ('V3', '*'), ('M5', 'v'), ('V4', '*'), ('M6', 'h'), ('V5', '*'), ('M7', 'v'), ('V6', '*'), ('M8', 'h'), ('V7', '*'), ('M9', 'v'), ('V8', '*'), ('M10', 'h'), ('V9', '*'), ('M11', 'v'), ('V10', '*'), ('M12', 'h'), ('V11', '*'), ('M13', 'v'), ('V12', '*'), ('M14', 'h'), ('V13', '*'), ('M15', 'v'), ('V14', '*'), ('Poly', 'v'), ('Fin', 'h'), ('Active', 'h'), ('Rvt', 'h'), ('Lvt', 'h'), ('Hvt', 'h'), ('Slvt', 'h'), ('Lisd', 'v'), ('Pc', 'h'), ('Pb', 'h'), ('V0', '*')])
align.cell_fabric.remove_duplicates WARNING : Layer Cboundary not in OrderedDict([('M1', 'v'), ('M2', 'h'), ('V1', '*'), ('M3', 'v'), ('V2', '*'), ('M4', 'h'), ('V3', '*'), ('M5', 'v'), ('V4', '*'), ('M6', 'h'), ('V5', '*'), ('M7', 'v'), ('V6', '*'), ('M8', 'h'), ('V7', '*'), ('M9', 'v'), ('V8', '*'), ('M10', 'h'), [... cut ...]
align.pnr.main INFO : OUTPUT error file at /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/3_pnr/SWITCHED_CAPACITOR_FILTER_0.errors
align.pnr.main INFO : OUTPUT gds.json /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/3_pnr/SWITCHED_CAPACITOR_FILTER_0.python.gds.json
Use KLayout to visualize the generated GDS: /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/SWITCHED_CAPACITOR_FILTER_0.gds
Use KLayout to visualize the python generated GDS: /home/smburns/DARPA/ALIGN-public/work/switched_capacitor_filter/SWITCHED_CAPACITOR_FILTER_0.python.gds
```

kkunal1408, 2022-03-01:
@stevenmburns The missing pin error can be waived:
This is because one aspect ratio of SCM_NMOS is too small to generate that pin. We have another SCM_NMOS with a different aspect-ratio and correct pins which get used during PnR.
@arvuce22 please correct me if you have anything else to add or can you fix the code to not generate this wrong primitive?

stevenmburns, 2022-03-01:
@arvuce22 We can really generate a wrong primitive. A primitive needs to have the pins is supposed to have. If it is ill formed in some way, we need to remove it before when generate a LEF file.

stevenmburns, 2022-03-02:
@Lastdayends Can you look at the routing open?

Lastdayends, 2022-03-02:
@stevenmburns Will look into the case.

stevenmburns, 2022-03-07:
@arvuce22 Any progress on removing non-fully formed primitives from consideration early in the flow. This switched_capacitor_filter example provides a good these case.

stevenmburns, 2022-03-07:
@Lastdayends Any progress on this switched_capacitor_filter routing failure.

arvuce22, 2022-03-07:
@stevenmburns I will fix it by the end of this week.

Lastdayends, 2022-03-07:
@stevenmburns 

The open is due to the error of missing pin in the lef file. Once the error is fixed. The open should be fixed.

stevenmburns, 2022-03-07:
Which pin is missing? Is it missing from the LEF file corresponding to the CC capacitor or is it something else?

Lastdayends, 2022-03-07:
@stevenmburns 

Block X_M4_M5 pin DB not found in lef file. It should be a pin in the primitive.

arvuce22, 2022-03-15:
@Lastdayends I have resolved the pin issues in #1021. Can you check this?

Lastdayends, 2022-03-15:
@arvuce22 

Will check it.

stevenmburns, 2022-11-16:
Seems to be resolved in #1218

closed 2022-11-16 by stevenmburns