ALIGN-analoglayout/ALIGN-public #1030: Improving Floorplan Efficiency

soneryaldiz, 2022-03-17:
```java
{
  "constraint": "Floorplan",
  "regions": [
    [ "A1", ...  "Ax" ],  // x instances
    [ "B1", ...  "By" ],  // y instances
    [ "C1", ...  "Cz" ],  // z instances
  ]
}
```
A floorplan constraint as shown above creates `x*y + y*z` vertical ordering constraints:
```java
{"constraint": "Order", "direction": "V", "instances": ["Ai", "Bj"]}
...
{"constraint": "Order", "direction": "V", "instances": ["Bi", "Cj"]}
...
}
```

The same intention could be specified with much fewer constraints if we could introduce two auxiliary variables that represent two horizontal lines. For example, a new constraint can be used to define these variables:
```java
{
  "constraint": "SpecifyCutlines",
  "direction": "H",
  "lines": ["k1", "k2"]
}
```
and the number of constraints reduce from `x*y + y*z` to `x + 2*y + z`:

```java
{"constraint": "Order", "direction": "V", "instances": ["Ai", "k1"]}
...
{"constraint": "Order", "direction": "H", "instances": ["k1", "Bi"]}
...
{"constraint": "Order", "direction": "H", "instances": ["Bi", "k2"]}
...
{"constraint": "Order", "direction": "H", "instances": ["k2", "Ci"]}
...
}
```
This cut line constraint can later be used to specify vertical lines as well.

@srini229 and @854768750 , thoughts? Would the suggestion above improve the efficiency of the placement algorithm?

srini229, 2022-03-18:
@soneryaldiz We don’t need to translate the floor plan constraint to pairs of ordering constraint if we come up with a constructive way to generate sequence pairs that honor ordering constraints. I can work on the constructive approach.

srini229, 2022-03-18:
It would indirectly implement the cut line constraint.

soneryaldiz, 2022-03-22:
@srini229 , here is more motivation for explicitly handling floorplan constraints, or at least,  checking for existence of blocks that are above (left) of all other blocks:

Suppose that n blocks are broken into 3 regions r1, r2 and r3 with n1, n2 and n3 blocks in each region respectively (i. e. n1 + n2 + n3 = n). These vertical regions basically imply vertical ordering constraints where each block a in ri is above each block in rj where j>i. 

Such a constraint implies the following structure on the positive and negative sequence pairs:
`(< r1 >)(< r2 >)(< r3 >) , (< r3 >)(< r2 >)(< r1 >) `
Consequently, the number of feasible solutions reduces to n1!^2 * n2!^2 * n3!^2. This representation also satisfies one or more vertical line of symmetric placement constraints that involve any blocks x and y where x != y and x and y are in different regions are in different regions. 

Example#1: 10 blocks organized in three regions of 1, 9 (basically one block is above all else)
10!^2 ~ 1.3e13   vs 9!^2 ~ 1.3e12 => An order of magnitude reduction

Example#2: 10 blocks organized in two equal-sized regions
10!^2 ~ 1.3e13   vs 5!^2 * 5!^2  ~ 2.1e8 => 5 orders of magnitude reduction

Example#3: 10 blocks organized in three regions of 2, 3, 5
10!^2 ~ 1.3e13   vs 2!^2 * 3!^2 * 5!^2 ~ 2.1e6 => 7 orders of magnitude reduction

Example#4: 19 blocks organized in three regions of 1, 5, 4, 6, 2, 1 (this is a production two stage amplifier circuit)
19!^2 ~ 1.5e34   vs vs 5!^2 * 4!^2 * 6!^2 * 2!^2 ~ 1.7e13 => 21 orders of magnitude reduction

Although the reduced spaces are still too large to enumerate, there could be other constraints (align, order, symmetry) that could further reduce the search space in each region.

soneryaldiz, 2022-03-22:
When the (horizontal) order flag is set for a floorplan constraint, the search space reduces to a single feasible sequence pair. For example, ABCDE, DECAB is the only sequence pair that satisfies the constraint:
```javascript
{
  “constraint”: “Floorplan”, 
  “regions”: [[A,B], [C], [D,E]],
  “order”: True
}
```

srini229, 2022-03-23:
@soneryaldiz I'll work on integrating this into the enumerative placer. It should be straightforward without the symmetry constraints. The topological sorting routine can be modified to accommodate this constraint.

soneryaldiz, 2022-03-23:
@srini229 , let me know if you want to talk about this briefly. The key observation: When there is a floorplan constraint that includes all instances in a circuit, then the sequence pair search space reduces significantly and this reduces subspace can be explored (perturbed) more efficiently. This constraint could easily be applied to most analog circuits by simply defining two regions: all pmos and all nmos.

soneryaldiz, 2022-03-24:
Notes from discussion:

- [ ] Retire KeepOrdering
- [ ] Construct an initial feasible solution using the topologically sorted H and V constraint graphs
- [ ] Refactor Perturbation choices to be ordering aware

854768750, 2022-03-24:
> @srini229 , let me know if you want to talk about this briefly. The key observation: When there is a floorplan constraint that includes all instances in a circuit, then the sequence pair search space reduces significantly and this reduces subspace can be explored (perturbed) more efficiently. This constraint could easily be applied to most analog circuits by simply defining two regions: all pmos and all nmos. Such a floorplan reduces the horizontal padding in each primitive and can also enables sharing taps.

If there is a floorplan constraint that includes all instances in a circuit, does this mean there will be only one solution? Because the order of rows is determined and the order inside each row is also determined.

soneryaldiz, 2022-11-03:
This is partially completed by #1180.

closed 2022-11-03 by soneryaldiz