ALIGN-analoglayout/ALIGN-public #1037: Runtime Improvements are needed

stevenmburns, 2022-03-24:
I'm using telescopic_ota as a baseline.
Here is a flame graph from vtune:
[image]

@soneryaldiz @srini229 @854768750 A lot of time is being spent in the ILP within the placer. There is some time wasted in spdlog that I will try to fix.

soneryaldiz, 2022-03-24:
This is very insightful, thank you! `spdlog::logger` is also non-negligible.

srini229, 2022-03-24:
@stevenmburns I'm looking into the ILP solver to see for better build options that we can utilize. I'm also looking at other solvers by the same group. 
For spdlog inside SeqPair::PerturbationNew, [these lines](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/6f5665c8cd58421275dd359c624be07ecc03263f/PlaceRouteHierFlow/placer/SeqPair.cpp#L1270-L1274) in SeqPair.cpp are most likely the ones affecting runtime. 

This can be fixed by moving these statements under a `if (spdlog::get_level() == spdlog::debug)`.

closed 2023-02-09 by soneryaldiz