ALIGN-analoglayout/ALIGN-public #1041: Master currently failing CI

stevenmburns, 2022-03-28:
This seems to be because of a test in `tests/pdk/finfet_pdk/test_placement_grid.py` taking a long time. I noticed this first by carefully examining the CI log which showed that this test started but didn't finish in a case where CI timedout.
To duplicate, try:
```
pip install -v -e .[test] --no-build-isolation
pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=RelWithDebInfo' --install-option='-DBUILD_TESTING=ON'
pytest --durations=10 -n 1 -vv -- tests/pdk/finfet_pdk/test_placement_grid.py
```
I'm getting:
```
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ pytest --durations=10 -n 1 -vv -- tests/pdk/finfet_pdk/test_placement_grid.py
Test session starts (platform: linux, Python 3.8.13, pytest 6.2.4, pytest-sugar 0.9.4)
cachedir: .pytest_cache
rootdir: /home/smburns/DARPA/ALIGN-public, configfile: pytest.ini
plugins: cov-2.12.1, flaky-3.7.0, dash-2.0.0, forked-1.3.0, timeout-1.4.2, cpp-2.0.0, mock-3.6.1, xdist-2.2.1, sugar-0.9.4
[gw0] linux Python 3.8.13 cwd: /home/smburns/DARPA/ALIGN-public
[gw0] Python 3.8.13 (default, Mar 16 2022, 17:28:59)  -- [GCC 7.5.0]
gw0 [5]
scheduling tests via LoadScheduling

 tests/pdk/finfet_pdk/test_placement_grid.py::test_scalings ✓                                           20% ██        
[gw0] PASSED tests/pdk/finfet_pdk/test_placement_grid.py 
 tests/pdk/finfet_pdk/test_placement_grid.py::test_check_metadata ✓                                     40% ████      
[gw0] PASSED tests/pdk/finfet_pdk/test_placement_grid.py 
 tests/pdk/finfet_pdk/test_placement_grid.py::test_ota_on_grid_h ✓                                      60% ██████    
[gw0] PASSED tests/pdk/finfet_pdk/test_placement_grid.py 
 tests/pdk/finfet_pdk/test_placement_grid.py::test_ota_on_grid_v ✓                                      80% ████████  
[gw0] PASSED tests/pdk/finfet_pdk/test_placement_grid.py 
 tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid ✓                                       100% ██████████
[gw0] PASSED tests/pdk/finfet_pdk/test_placement_grid.py 

--------------------------- generated xml file: /home/smburns/DARPA/ALIGN-public/junit.xml ---------------------------
================================================ slowest 10 durations ================================================
450.11s call     tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid
2.87s call     tests/pdk/finfet_pdk/test_placement_grid.py::test_ota_on_grid_h
2.80s call     tests/pdk/finfet_pdk/test_placement_grid.py::test_ota_on_grid_v
0.34s call     tests/pdk/finfet_pdk/test_placement_grid.py::test_scalings
0.24s call     tests/pdk/finfet_pdk/test_placement_grid.py::test_check_metadata
0.00s setup    tests/pdk/finfet_pdk/test_placement_grid.py::test_scalings
0.00s setup    tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid
0.00s setup    tests/pdk/finfet_pdk/test_placement_grid.py::test_ota_on_grid_h
0.00s setup    tests/pdk/finfet_pdk/test_placement_grid.py::test_check_metadata
0.00s setup    tests/pdk/finfet_pdk/test_placement_grid.py::test_ota_on_grid_v

Results (458.31s):
       5 passed
```
This is likely to be slower if we run the test in`Debug` mode.

stevenmburns, 2022-03-28:
@soneryaldiz What does this test do and can we make it take less time?

stevenmburns, 2022-03-28:
Here is another run that took a 100 less seconds:
```
(.venv) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ pytest --durations=10 -n 1 -vv tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid
Test session starts (platform: linux, Python 3.8.13, pytest 6.2.4, pytest-sugar 0.9.4)
cachedir: .pytest_cache
rootdir: /home/smburns/DARPA/ALIGN-public, configfile: pytest.ini
plugins: cov-2.12.1, flaky-3.7.0, dash-2.0.0, forked-1.3.0, timeout-1.4.2, cpp-2.0.0, mock-3.6.1, xdist-2.2.1, sugar-0.9.4
[gw0] linux Python 3.8.13 cwd: /home/smburns/DARPA/ALIGN-public
[gw0] Python 3.8.13 (default, Mar 16 2022, 17:28:59)  -- [GCC 7.5.0]
gw0 [1]
scheduling tests via LoadScheduling

 tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid ✓                                       100% ██████████
[gw0] PASSED tests/pdk/finfet_pdk/test_placement_grid.py 

--------------------------- generated xml file: /home/smburns/DARPA/ALIGN-public/junit.xml ---------------------------
================================================ slowest 10 durations ================================================
365.08s call     tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid
0.00s setup    tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid
0.00s teardown tests/pdk/finfet_pdk/test_placement_grid.py::test_cmp_on_grid

Results (367.02s):
       1 passed
```

soneryaldiz, 2022-03-28:
`test_cmp_on_grid` runs the flow on comparator with `PlaceOnGrid` constraints.

soneryaldiz, 2022-03-28:
What is the runtime if we stop the flow before routing?

stevenmburns, 2022-03-28:
Routing will take essentially zero time. This is all placement. I don't know about a logging contribution though. I'll run this through vtune.

stevenmburns, 2022-03-28:
Here is the flame graph from VTUNE
[image]
All the time is spent in the ILP calls (2 secs in Perturb New, 3 secs in routing, ~400 sec in total)

soneryaldiz, 2022-03-28:
Thanks @stevenmburns. I still prefer to keep this test as it is the only non-trivial test for PlaceOnGrid. I will try refactoring the test so that it does not take this long.

stevenmburns, 2022-03-28:
There seems to be no substantial logger component (at least in the branch I'm on. I'll start a PR.)

kkunal1408, 2022-11-03:
closing this issue as master is passing now.

closed 2022-11-03 by kkunal1408