ALIGN-analoglayout/ALIGN-public #1048: Create GDS from placement-verilog-json

kkunal1408, 2022-03-31:
While using ALIGN as multiple entry and exit point, running it till the placement only creates placement Verilog JSON. How to create a GDS from this file?

stevenmburns, 2022-03-31:
What do you want to do? 
The test https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/tests/flow/test_flow.py shows how to run the flow one step at a time.
You should be able to run up to placement:
```
mkdir -p work/telescopic_ota
cd work/telescopic_ota
schematic2layout.py ../../examples/telescopic_ota --flow_stop 3_pnr:place
```
and then restart it to generate routing (with GDS.)
```
schematic2layout.py ../../examples/telescopic_ota --flow_start 3_pnr:route
```

srini229, 2022-03-31:
@stevenmburns We added a script to generate GDS files from the placement verilog and leaf gds.json files. Please find the script in the attached branch. This will help eventually get rid of the .gds.json files. A final postprocess routine can generate the GDS files from the placement/routing json and leaf json files.

kkunal1408, 2022-03-31:
> What do you want to do? The test https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/tests/flow/test_flow.py shows how to run the flow one step at a time. You should be able to run up to [...]

It was a request from another user. He wanted to use placement GDS.

closed 2022-03-31 by srini229