ALIGN-analoglayout/ALIGN-public #1055: Floating Subcircuit Pins

soneryaldiz, 2022-04-06:
ALIGN flow throws an assertion if a subcircuit pin is not connected any instance within the subcircuit. 

1. What is the key reason?
2. Could ALIGN quietly ignore floating pins from the hierarchy?

soneryaldiz, 2022-04-06:
@kkunal1408 , if there is no significant reason, I can create a PR with a failing test to drive development.

soneryaldiz, 2022-04-06:
@kkunal1408 , what if these pins are removed from the hierarchy so that PnR is never exposed to these?

kkunal1408, 2022-04-06:
It can be done, but that is more like modifying the netlist and in our current assumption we assume the designer has provided a correct netlist.

soneryaldiz, 2022-04-06:
This is not really about correctness as a subcircuit definition with a floating pin is legal. This scenario happens on legacy schematics where a configuration pin that was used in the past is no longer needed. Removing this pin thru a deep hierarchy of cells is not a trivial task.

soneryaldiz, 2022-04-06:
@Lastdayends , could PnR tolerate floating pins? It already does tolerate floating nets in a subcircuit. Could you please let us know about the complexity?

kkunal1408, 2022-04-06:
Yes in a schematic it's legal to keep a floating(it shows a warning) or you can connect to "noconn" to get rid of the warning. What do designers do with that in the layout?
Do not connect in the lowermost hierarchy or also keep it floating in the hierarchies above it?
If removing the pin and all corresponding nets till the top hierarchy is allowed I can do the modifications to remove the pin.

Lastdayends, 2022-04-06:
@kkunal1408 @soneryaldiz 

Do you mean a pin defined as a port of a subcircuit, but no internal net connects to that pin? May I see what kink of error?

soneryaldiz, 2022-04-06:
I will start a PR shortly.

soneryaldiz, 2022-04-06:
Please see #1058

854768750, 2022-04-19:
This is solved in #1058, can this issue be closed?

closed 2022-04-19 by soneryaldiz